OctoPrint/OctoPrint #2617: API clients have trouble accessing job info

koenkooi, 2018-04-30:
#### What were you doing?

I was starting prints by various means, the lcd on the printer for an SD print, octoprint web interface for a serial print, octoclient for a serial print.
In all those cases octoclient wouldn't display job info. This started happening around the time the user info changes landed last week. The log is full of these:

```
Apr 30 08:31:25 raspberrypi2 octoprint[23585]: 2018-04-30 08:31:25,061 - tornado.access - ERROR - 500 GET /api/job (172.20.0.33) 24.02ms
Apr 30 08:31:28 raspberrypi2 octoprint[23585]: 2018-04-30 08:31:28,054 - octoprint - ERROR - Exception on /api/job [GET]
Apr 30 08:31:28 raspberrypi2 octoprint[23585]: Traceback (most recent call last):
Apr 30 08:31:28 raspberrypi2 octoprint[23585]:   File "/usr/lib/python2.7/site-packages/flask/app.py", line 1817, in wsgi_app
Apr 30 08:31:28 raspberrypi2 octoprint[23585]:     response = self.full_dispatch_request()
Apr 30 08:31:28 raspberrypi2 octoprint[23585]:   File "/usr/lib/python2.7/site-packages/flask/app.py", line 1477, in full_dispatch_request
Apr 30 08:31:28 raspberrypi2 octoprint[23585]:     rv = self.handle_user_exception(e)
Apr 30 08:31:28 raspberrypi2 octoprint[23585]:   File "/usr/lib/python2.7/site-packages/flask/app.py", line 1381, in handle_user_exception
Apr 30 08:31:28 raspberrypi2 octoprint[23585]:     reraise(exc_type, exc_value, tb)
Apr 30 08:31:28 raspberrypi2 octoprint[23585]:   File "/usr/lib/python2.7/site-packages/flask/app.py", line 1475, in full_dispatch_request
Apr 30 08:31:28 raspberrypi2 octoprint[23585]:     rv = self.dispatch_request()
Apr 30 08:31:28 raspberrypi2 octoprint[23585]:   File "/usr/lib/python2.7/site-packages/flask/app.py", line 1461, in dispatch_request
Apr 30 08:31:28 raspberrypi2 octoprint[23585]:     return self.view_functions[rule.endpoint](**req.view_args)
Apr 30 08:31:28 raspberrypi2 octoprint[23585]:   File "/usr/lib/python2.7/site-packages/octoprint/server/api/job.py", line 73, in jobState
Apr 30 08:31:28 raspberrypi2 octoprint[23585]:     "printingUser": currentData["printingUser"],
Apr 30 08:31:28 raspberrypi2 octoprint[23585]: KeyError: 'printingUser'
```

#### What did you expect to happen?

I expected the client to show at least the percent complete and time elapsed. The big "Not Printing" text in octoclient  while SD printing is an existing bug in octoclient.

#### What happened instead?

Only 'Printing from SD' shows

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

_maintenance_  revision 4bbe49a

#### Operating System running OctoPrint

```
root@raspberrypi2:~# cat /etc/os-release 
VERSION_ID="v2017.12"
NAME="Angstrom"
ID="angstrom"
PRETTY_NAME="The Ångström Distribution v2017.12"
ANSI_COLOR="1;35"
HOME_URL="http://www.angstrom-distribution.org"
```

#### Printer model & used firmware incl. version

1. Prusa i3 MK3 running https://github.com/prusa3d/Prusa-Firmware/commit/e2b3553aee9cd16f22d66e4f748ec16b5147c6e4
2. Fabrikator mini V1 running https://github.com/MarlinFirmware/Marlin/commit/50cc55d172b04db3a584319cfd5b84e34bda4344

#### Browser and version of browser, operating system running browser

OSX/safari

#### Link to octoprint.log

https://gist.github.com/koenkooi/82fefc15516622cf560a300e07602114

#### Screenshot(s)/video(s) showing the problem:

[image]
[image]

I have read the FAQ.

hashashin, 2018-05-01:
Hi @koenkooi, The error in my logs looks the same: https://git.io/vpEQJ
Removing the offending line "works for me"<sup>®️</sup> as workaround, until proper fix it's made:
```
diff --git a/src/octoprint/server/api/job.py b/src/octoprint/server/api/job.py
index 909b74f4..a4813011 100644
--- a/src/octoprint/server/api/job.py
+++ b/src/octoprint/server/api/job.py
@@ -70,5 +70,4 @@ def jobState():
                "job": currentData["job"],
                "progress": currentData["progress"],
                "state": currentData["state"]["text"],
-               "printingUser": currentData["printingUser"],
        })
```

Salud.

foosel, 2018-05-03:
That was something that slipped through after a refactoring. Fixed now by the above commit.

commit 9addbbb9 (2018-05-03): Fix refactoring bug

Closes #2617
closed 2018-05-03 by foosel