OctoPrint/OctoPrint #2618: GCode Viewer Inverted to what the printer prints

TimothyGold, 2018-04-30:
Not a huge issue but just thought I would mention this. I use a Prusa MK3 and i need to invert control on the y axis. I think this may be why the GCode Viewer shows an inverted view of the print on the bed to what it is really doing on the printer.

foosel, 2018-05-03:
> May be caused by the fact that the Y axis for the Prusa MK3 controls needs to be inverted from default.

I have a MK3 myself. I don't have to invert anything.

Check if the issue persists in safe mode. If it does, please provide a screenshot of the issue, a file + printer profile settings with which to reproduce it plus **the requested logs**.

PS:

> None needed

^- best way to *not* get a problem is resolved to decide on your own what kind of information is "not needed". It's requested for a reason. You could have saved us both some time (which at least I could have used to maybe solve the issue...) if you'd provided that stuff with your ticket from the get go.

foosel, 2018-05-08:
I still don't get why you think you need to invert your Y axis on your MK3. 

The up arrow increases the *Y coordinate of the print head*. On a machine with a moving bed that means the bed comes *forward* but relative to the bed the print head moves *backward* in direction of increasing Y.

This is a MK3 with stock firmware and no inversion checkbox applied:

[image]

Does your's behave differently?

To look further into this I also still need a) a test in safe mode, b) a file and c) your printer profile.

foosel, 2018-05-22:
Ping?

TimothyGold, 2018-05-22:
@foosel Thank you for the clarification and yes it is actually working as designed.  Looking at the bed from my camera point of view was confusing me but looking at the controls from the point of view increasing and decreasing the actual Y axis is working as intended.  This issue can be close...  Thanks again for the help...

closed 2018-05-22 by TimothyGold