OctoPrint/OctoPrint #2619: [BUG] Fake ACK will cancel ongoing print

bolsoncerrado, 2018-04-30:
#### What were you doing?

Monitoring ongoing prints on several printers, I accidentally click FAKE ACK on the wrong printer.
The screen was showing 

State: Cancelling...
File: mmupet~1.gco (SD)

#### What did you expect to happen?

I wasnt expecting the ongoing print to get cancelled anyway!

#### What happened instead?

It cancelled the ongoing "Print from SD" on the printer I clicked FAKE ACK.

#### Did the same happen when running OctoPrint in safe mode?

Didnt try.

#### Version of OctoPrint

OctoPrint 1.3.8 running on OctoPi 0.13.0

#### Operating System running OctoPrint

OctoPi 13 Raspbian Jessie

#### Printer model & used firmware incl. version

Prusa MK3

#### Browser and version of browser, operating system running browser

From cell phone with latest Android & Chrome

#### Link to octoprint.log

https://pastebin.com/N3QGedCY

#### Link to contents of terminal tab or serial.log

Already overwritten.
It was showing "echo: busy" all the way

I have read the FAQ.

foosel, 2018-05-02:
I can't do a thing here without a `serial.log`.

bolsoncerrado, 2018-05-02:
Seems I had it disabled :'(

foosel, 2018-05-02:
Then enable it and try to reproduce ;)

foosel, 2018-05-22:
I'm still waiting for that `serial.log`.

closed 2018-06-07 by foosel