OctoPrint/OctoPrint #2621: Render Undrendered Timelapse not pulling fps from config

gingerbreadassassin, 2018-05-02:
#### What were you doing?

Trying to render an unrendered timelapse. I have offloaded this rendering to a more powerful server via a nfs share, roughly following [this](http://crankycoder.net/2016/11/07/no-i-want-to-print-more-or-offloading-octoprint-timelapse-videos/).

#### What did you expect to happen?
I expected the ffpmeg arguments to reflect my target fps setting (60)

#### What happened instead?
The "-framerate" parameter kept using the default value of 25

#### Did the same happen when running OctoPrint in safe mode?

I didn't try. The only plugins I have installed are fullscreen and octorant, and I think I found the problem anyway.

#### Version of OctoPrint

OctoPrint 1.3.8 

#### Operating System running OctoPrint

OctoPi 0.15.0PE

#### Printer model & used firmware incl. version

Prusa i3 MK3

#### Browser and version of browser, operating system running browser

Chrome Version 65.0.3325.181 (Official Build) (64-bit)

#### Link to octoprint.log

[octoprint.log](https://gist.github.com/gingerbreadassassin/744dd22da129ca76a1db962fc50c8291)

#### Link to contents of terminal tab or serial.log

Not an issue with the printer.

#### Link to contents of Javascript console in the browser

I don't believe this to be a browser issue.

#### Screenshot(s)/video(s) showing the problem:

This is unavailable, as I have "fixed" the problem, and the tmp folder has been cleared out. I'm clicking on the camera button under the "unrendered timelapses" section.

My fix was to edit [this line](https://github.com/foosel/OctoPrint/blob/6d7f8a2a64b364504f11dafd5009a85fb785127a/src/octoprint/timelapse.py#L157) so that fps=60. I added logging calls to troubleshoot, which show up with their line numbers in the octoprint.log file as warnings.

I believe [this line](https://github.com/foosel/OctoPrint/blob/6d7f8a2a64b364504f11dafd5009a85fb785127a/src/octoprint/static/js/app/viewmodels/timelapse.js#L363) should be passing the fps parameter to that function, but I'm not well versed in JS to try it.

I have read the FAQ.

gingerbreadassassin, 2018-05-02:
For what it's worth, the bitrate parameter functions as expected.

foosel, 2018-05-03:
The backend code was at fault, not the frontend code. Fixed now on `maintenance`, soon `devel` and to be released with 1.3.9.

commit 13b85fbb (2018-05-03): Fix fps source when rendering unrendered timelapse

Should be taken from config unless provided as parameter, not hard coded
to 25.

Closes #2621
closed 2018-07-25 by foosel