OctoPrint/OctoPrint #2625: Package metadata of plugins installed through pip 10+ not found by OctoPrint

foosel, 2018-05-04:
### Problem

Starting with pip 10, a `pip install` for a VCS URL (which apparently also includes https zip file URLs) [will now build a wheel](https://pip.pypa.io/en/stable/news/#id5) of the installed URL via `python setup.py bdist_wheel` and install that instead of running `python setup.py install`. That means that instead of dumping the package metadata into an `egg-info/PKG-INFO` it will now be dumped into `dist-info/METADATA` (see PEP 376).

OctoPrint's current way of retrieves a plugin package's metadata when going through the plugin entry point results involves [a custom implementation of `pkginfo.Installed`](https://github.com/foosel/OctoPrint/blob/c55c568bbddf5112facf6aa3a173d500936b6cbf/src/octoprint/plugin/core.py#L1594-L1642) which takes the entry point object and tries to find the correct metadata file from that based on some path traversing. This only includes looking at `.egg-info` files and directories though and hence no metadata will be found for plugins installed from wheels. 

The result of that is a lack of metadata normally retrieved from the package data (such as license, homepage URL, author and such) and also an inability to detect what plugin was installed during finalization of the plugin manager's install routine, causing a warning popup.

Plugins continue to function just fine though, it's just the metadata that doesn't get read properly. Hence this should be classified as non critical bug.

### Solution

OctoPrint's plugin manager needs to be adjusted so it can also fetch metadata for packages installed from wheels.

A first throw at this has been done with c3eebdb158ce3f409c3aa56d8c84f7a39512b45a, however this seems overly complicated on second glance considering we have access to a `Distribution` object on the entry point object itself. A second throw at simplifying this will be made shortly.

foosel, 2018-05-04:
Open question: Why can't the plugin manager detect the plugin on install? The lack of metadata shouldn't be causing this and that smells of a bug in hiding.

ntoff, 2018-05-04:
I assume this is why I got a popup "installation of: *unknown* was successful, however a restart of octoprint is necessary" when installing from a zip file?

foosel, 2018-05-05:
Yep, exactly.

ntoff, 2018-05-05:
1.3.8 release on my custom install on linux mint

foosel, 2018-07-25:
1.3.9 has just been released.

closed 2018-07-25 by foosel