OctoPrint/OctoPrint #2627: text in octopi-wpa-supplicant.txt is wrong. wep_key0= <cannot have double quotes>

tjsoco, 2018-05-05:
I'm using version 1.3.8.

I'm using WEP on my wireless network. 
The section in octopi-wpa-supplicant.txt
looked like this (I think it looked like this, I have since edited it).
network={
  ssid="put SSID here"
  key_mgmt=NONE
  wep_key0="put password here"
  wep_tx_keyidx=0
}
But the "wep_key0 is a 26 hexadecimal string.
It doesn't work if it is bracketed by double quotes.
When I put it in double quotes, the WiFi didn't work and I found an error log in the syslog that said key was too long.
Once I removed the double quotes, WiFi worked fine.

I think the correct octopi-wpa-supplicant.txt line should be changed from:
wep_key0="put password here"
to:
wep_key0=26_hexadecimal_character_key_goes_here

Yes, I know I should not be using WEP.
Thanks for Octopi!

ntoff, 2018-05-05:
In that case you might want the OctoPi bug tracker https://github.com/guysoft/OctoPi/ as this tracker is only for OctoPrint.

foosel, 2018-05-05:
Yep, wrong bug tracker.

closed 2018-05-05 by foosel