OctoPrint/OctoPrint #2629: After slicing for single extruder temperature gcode is also added for 2nd extruder

StevenBruinen, 2018-05-05:
#### What were you doing?

I've setup my printer profile for dual extruders. But after slicing a file for only one extruder, with onboard Cura, it results in gcode with temperature settings also for the 2nd extruder.

1. Setup printer profile for dual extrusion.
2. Slice a STL file with the Cura plugin.
3. Check starting gcode for M104 / M109 T0 & T1 settings

#### What did you expect to happen?

Only temperature settings for the 1st extruder should be added to the gcode.

#### What happened instead?

```
;Generated from xyzCalibration_cube.tmp.1525278632258.stl505ccd9b89060a2383faaf0533ce02b27524a196
;Generated with Cura_SteamEngine 15.04.6

M140 S50.0
M104 T1 S220.0
M109 T0 S220.0
M109 T1 S220.0
T0
M190 S50.0
[...] 
M104 T0 S?print_temperature
```

#### Did the same happen when running OctoPrint in safe mode?
Can't test.

Because the Cura addon isn't available in safe mode

#### Version of OctoPrint
OctoPrint 1.3.8

#### Operating System running OctoPrint
OctoPi 0.14.0

#### Printer model & used firmware incl. version
Custom build CoreXY running Marlin 1.1.x

#### Browser and version of browser, operating system running browser
Chrome v66 running on Windows 10

#### Link to octoprint.log

[octoprint.log](https://github.com/foosel/OctoPrint/files/1976539/octoprint.log)

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-06-18:
> **Did the same happen when running OctoPrint in safe mode?**
>
> Can't test.
>
> Because the Cura addon isn't available in safe mode

Just for the record: It actually is available. It's a bundled plugin, those *aren't* disabled in safe mode.

Could you share your slicing and printer profiles so I can reproduce this?

StevenBruinen, 2018-06-20:
I can't see the Cura tab when running in safe mode so thats why I thought it wasn't available:

[image]

But here are my settings for the printer and slicing profile:

[image]
[image]
[image]
[image]

[Cura_PLA.ini.txt](https://github.com/foosel/OctoPrint/files/2118497/Cura_PLA.ini.txt)

foosel, 2018-06-20:
Ah. Let me guess, you are using the "Full Featured Slicer" plugin and *that's* what you missed when running in safe mode. OctoPrint's built-in Cura plugin doesn't provide a tab, however the aforementioned third party plugin using it does.

Thanks for your profiles, I'll take a look and see if I can reproduce this behaviour.

foosel, 2018-10-29:
Found and fixed via dafa4c9. I also noticed that your provided profile had some issues with its placeholders that I've seen a couple of times now, probably caused by some Cura export oddities. I've added a workaround for it in 4eb81aa.

Both commits are available on `maintenance`, soon `devel` and will be part of 1.3.10.

commit dafa4c97 (2018-10-29): cura: fix extruder count for start gcode selection

Fixes #2629
closed 2018-12-10 by foosel