OctoPrint/OctoPrint #2631: Unable to start instance

sanchosk, 2018-05-06:
#### What were you doing?

1. Updated octo using git pull to latest devel version
2. run ./venv/bin/pip install pip --upgrade
3. run ./venv/bin/python setup.py clean
4. run ./venv/bin/python setup.py install
5. run ./venv/bin/octoprint serve

#### What did you expect to happen?

I expected octo to start :)

#### What happened instead?

Error:
2018-05-06 15:16:52,328 - octoprint.cli.server - ERROR - Uncaught exception
Traceback (most recent call last):
  File "./venv/bin/octoprint", line 11, in <module>
    load_entry_point('OctoPrint==1.4.0.dev1190+gd8eb5754', 'console_scripts', 'octoprint')()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.4.0.dev1190+gd8eb5754-py2.7.egg/octoprint/__init__.py", line 565, in main
    octo(args=args, prog_name="octoprint", auto_envvar_prefix="OCTOPRINT")
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/click-6.7-py2.7.egg/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/click-6.7-py2.7.egg/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/click-6.7-py2.7.egg/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/click-6.7-py2.7.egg/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/click-6.7-py2.7.egg/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/click-6.7-py2.7.egg/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.4.0.dev1190+gd8eb5754-py2.7.egg/octoprint/cli/server.py", line 183, in serve_command
    ignore_blacklist)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.4.0.dev1190+gd8eb5754-py2.7.egg/octoprint/cli/server.py", line 104, in run_server
    octoprint_server.run()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.4.0.dev1190+gd8eb5754-py2.7.egg/octoprint/server/__init__.py", line 279, in run
    self._start_intermediary_server()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.4.0.dev1190+gd8eb5754-py2.7.egg/octoprint/server/__init__.py", line 1627, in _start_intermediary_server
    bind_and_activate=False)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.4.0.dev1190+gd8eb5754-py2.7.egg/octoprint/server/__init__.py", line 1611, in __init__
    HTTPServer.__init__(self, *args, **kwargs)
  File "/usr/lib/python2.7/SocketServer.py", line 414, in __init__
    self.socket_type)
  File "/usr/lib/python2.7/socket.py", line 191, in __init__
    _sock = _realsocket(family, type, proto)
error: [Errno 97] Address family not supported by protocol

#### Did the same happen when running OctoPrint in safe mode?

Yes. I also tried to clean the .octoprint, wipe the git repo, re-download from scratch and attempt "first run", the result is the same.

#### Version of OctoPrint

1.4.0.dev1190+gd8eb5754
#### Operating System running OctoPrint

Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.4 (stretch)
Release:	9.4
Codename:	stretch

#### Printer model & used firmware incl. version

Tronxy X5S with RAMPS instead of native board.

#### Browser and version of browser, operating system running browser

N/A

#### Link to octoprint.log

https://gist.github.com/sanchosk/fab06dd84adfe68eb8c43697ff6e987e

#### Link to contents of terminal tab or serial.log

https://gist.github.com/sanchosk/0aa3149408d286c0d0b8b78ca7d8997d

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s)/video(s) showing the problem:

N/A

I have read the FAQ.

Sure :)

sanchosk, 2018-05-06:
Checking master branch - setup clean and install and Octo starts again.

foosel, 2018-05-08:
What about the `maintenance` branch?

sanchosk, 2018-05-08:
Maintenance starts OK no problem.

sanchosk, 2018-05-08:
I did another pull (no update), re-checked to devel and it started no problem.
Perhaps some local network issue or something, closing the ticket.

closed 2018-05-08 by sanchosk