OctoPrint/OctoPrint #2632: Serial connection loop crash

spaceharrier, 2018-05-06:
#### What were you doing?

1. Routine print in ABS, Prusa i3 Mk2s
2. Error did not recur when retrying same Gcode file
3. ...

#### What did you expect to happen?

Successful print

#### What happened instead?

Aborted print-in-motion, print head left mid-extrude.

#### Did the same happen when running OctoPrint in safe mode?

Didn't try safe mode, as subsequent attempt to print same file in regular mode was successful. Filing a bug because inspecting the log file I came across a line saying "please report this in OctoPrint's bug tracker" :)

#### Version of OctoPrint

1.3.8

#### Operating System running OctoPrint

Linux 3d 4.14.37-v7+ #1111 SMP Thu Apr 26 13:56:59 BST 2018 armv7l GNU/Linux

#### Printer model & used firmware incl. version

Prusa i3 Mk2s, firmware v3.1.0

#### Browser and version of browser, operating system running browser

n/a

#### Link to octoprint.log

https://pastebin.com/uifPKqDw

#### Link to contents of terminal tab or serial.log

Serial log not enabled. Nothing in terminal -- error seen when browsing from phone to check on print.

#### Link to contents of Javascript console in the browser

n/a - didn't capture anything (Chrome on Android)

#### Screenshot(s)/video(s) showing the problem:

n/a 

I have read the FAQ.

foosel, 2018-05-08:
Thanks for reporting this, looks like a race condition.

Btw, your log is filled to the brim with resend requests, which is no surprise considering you are still running firmware 3.1.0 [which is known to cause severe communication issues](https://discourse.octoprint.org/t/printer-specific-original-prusa/239#headline--communication-issues-3.1.0). You should *really really* update your firmware, you are risking failed prints with this firmware version.

foosel, 2018-05-30:
You definitely still need an updated firmware though, I fixed the race condition inside OctoPrint, I can't fix the firmware running into endless communication issues due to an overwhelmed controller.

commit d124efdb (2018-06-05): Fix race condition in resend handling

Could be triggered if two resend requests without an ok were directly
following each other, causing a resend ok timer to be triggered for each
and a consequent attempt to process the same resend status again.

Should fix #2632
closed 2018-07-25 by foosel