OctoPrint/OctoPrint #2637: [Request] Make tab ids consistent with config.yaml appearance settings and plugin tabs

jneilliii, 2018-05-12:
While developing the TabOrder plugin I have run across an inconsistency with how the the li element ids are assigned between core tabs and plugin tabs.  It would make life a lot easier for developers if this was the same for all tabs.  It would be great for the names stored in config.yaml for order also match the same template.  For example, as it stands now we have the following:

**id** | **hash** | **config.yaml**
-- | ---- | -----------
`temp_link` | `#temp` | `temperature`
`term_link` | `#term` | `terminal`
`gcode_link` | `#gcode` | `gcodeviewer`
`tab_plugin_<pluginidentifier>_link` | `#tab_plugin_<pluginidentifier>` | `plugin_<pluginidentifier>`

Proposed change

**id** | **hash** | **config.yaml**
-- | ---- | -----------
`tab_temperature_link` | `#temperature` | `temperature`
`tab_terminal_link` | `#terminal` | `terminal`
`tab_gcodeviewer_link` | `#gcodeviewer` | `gcodeviewer`
`tab_plugin_<pluginidentifier>_link` | `#tab_plugin_<pluginidentifier>` | `plugin_<pluginidentifier>`

ntoff, 2018-05-12:
Does that extend to the side bar items as well?

foosel, 2018-06-12:
In principle I agree fully with fixing this inconsistency. What I'm worried here though is any kind of existing plugins that rely on the current IDs to be present (e.g. themeify, touchui and so on). Since an element can't have to IDs, we can't provide a fallback here. That makes this tricky from a backwards compatibility point of view.

jneilliii, 2018-06-12:
It's no problem Gina.  I have a workaround in place on my plugins already.  I agree backwards compatibility would be a nightmare on this one.

closed 2018-06-12 by jneilliii