OctoPrint/OctoPrint #2638: GCODE Scripts are not saving if power disconnects

vtulin, 2018-05-12:
I don't know if it's a programmed behavior, but when I click "Save" on the scripts tab, it doesn't save it in permanent memory, which in my opinion should, instead if you unplug the power and plug back you'll notice that your scripts are gone. If you do a normal reboot scripts are saving normally. If it's a normal behavior for a some reason, a line "You should reboot to save the scripts" would be perfect.
Octoprint 1.3.8.

ntoff, 2018-05-13:
Not a bug, just a byproduct of how linux filesystems work and why you ***MUST ALWAYS*** shutdown OctoPrint properly if running on a raspberry pi / linux based OS.

vtulin, 2018-05-13:
> you MUST ALWAYS shutdown OctoPrint properly if running on a raspberry pi / linux based OS
Why is that so? Is it operates differently from other Linux types? Yes I know that it's not a good way to turn it off by unpluging it, but are you saying that it is normal for file system to not write changes to disk unless it reboots?

hashashin, 2018-05-13:
Running "sync" in the command line before unplug the power, should workaround this. But it's always better do a proper shutdown/reboot. https://linux.die.net/man/8/sync

Salud.

John-Mc, 2018-05-13:
> are you saying that it is normal for file system to not write changes to disk unless it reboots?

Yes, although the key here is a proper shutdown, not a reboot (a proper shutdown is part of the reboot process, so you are covered there). The OS needs to take care of some final housekeeping items prior to disconnecting the power

vtulin, 2018-05-13:
I found out that the problem was not in the power, but in a UTF8 comments, the scripts are not saving if there are comments in my encoding. Would that be an issue?

foosel, 2018-05-14:
Please provide _all_ requested information. Without that information this ticket cannot be looked into and will be closed. Thanks.

closed 2018-05-27 by GitIssueBot