OctoPrint/OctoPrint #2641: "Update now" button not working

gege2b, 2018-05-14:
Hi there

#### What were you doing?

- Login on my instance and the "upgrade available" popup showed
- Click on the "update now" button
- ... Nothing. The popup is still displayed

No way to close the popup, except by clicking on "ignore"
Hard refresh (CTRL+F5) of the page => problem still occurs
Delete browser cache => probleme still occurs

#### What did you expect to happen?

Proceed to the update flawlessly

#### What happened instead?

Nothing, popup's "update now" button is not reacting, the "ignore" button closes the popup

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

OctoPrint 1.4.0.dev1190+gd8eb5754 

#### Operating System running OctoPrint

Raspbian GNU/Linux 9.4 (stretch) on raspberry pi3

#### Printer model & used firmware incl. version

Prusa mendel with Marlin 1.1.8 (but it's not related)

#### Browser and version of browser, operating system running browser

Firefox 59.0.3 64bits on Windows 10 on the first try, then I upgraded firefox to 60.0 with the same problem

#### Link to octoprint.log

[octoprint.log](https://framabin.org/?fefba96336d0497d#m67OLConYkgCCfg9fbEiTjcP/D9G66ExOSsBoo+QdVc=)

#### Link to contents of terminal tab or serial.log

I don't think it's needed because prints goes well

#### Link to contents of Javascript console in the browser
```
Less has finished and no sheets were loaded.
Champs mot de passe présents sur une page non sécurisée (http://). Cela représente un risque de sécurité permettant le vol d’identifiants de connexion.[En savoir plus] <=== my instance isn't using SSL so that's OK
Starting dependency resolution...
... dependency resolution done
Initial application setup done, connecting to server...
Connected to the server
Finalizing application startup
Going to bind 32 view models...
Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate since it does not exist
User gege logged in
... binding done
Application startup complete
```

#### Screenshot(s)/video(s) showing the problem:

Nothing fancy here : 
[image]

I have read the FAQ.

gege2b, 2018-05-14:
Additional info : Given the changes in user permissions granularity, I also tried to logout/login to eventually fetch new permissions (my user is admin)

I also tried to give "Users" group the update privilege (temporarly)

they both didn't work

gege2b, 2018-05-15:
Additional info² : Tested with microsoft Edge and the problem is the same

foosel, 2018-05-15:
That was a left-over from a refactoring. A permissions check was simply checking for a non existing (since renamed) permission. Should be fixed by the above commit (and another such issue that prevented the log dialog from opening should be fixed by 7d40320a5b05983b2114df2f0ed79d3b29cd2826).

closed 2018-05-15 by foosel

gege2b, 2018-05-15:
oh great ! thank you

I guess I should do the update manually then ? (I've not access to my instance right now so I can't test)

foosel, 2018-05-16:
Yeah, that would be best. 

I mean, you *could* also hand craft the update request in the JS console (the issue was purely client side) but that would be kinda hackish ;) In case you want to try it anyhow, open the console with F12, then paste this and hit enter:

```
OctoPrint.plugins.softwareupdate.updateAll()
```

That will work work around the broken button and should give you the "update in progress" notification. It won't show the log dialog though, due to a similar issue with that (also fixed in the current commit).

commit 09456b92 (2018-05-15): SWU: Fix update button not working

Closes #2641