OctoPrint/OctoPrint #2643: Can't switch to hidden tabs

gege2b, 2018-05-14:
Hi there, again

#### What were you doing?
- trying to access a hidden tab (Octolapse in this case, but could be anyone)
- click on the three-bars menu on the right corner and on octolapse tab
- the octolapse panel show up really quickly and then disapear
- tried to click on the "octolapse" button in the navbar, idem
- Reduced the page zoom by some percent so the tab is displayed on the tab bar
- tried again and the panel is finally displayed

#### What did you expect to happen?
The panel to show up normally

#### What happened instead?
Look at this screen cap, more useful than a long text
[octoprint2.zip](https://github.com/foosel/OctoPrint/files/2000860/octoprint2.zip)

Note : I think the problem appeared on a recent commit (not older than a month) because all was fine when I installed octolapse at the end of march

#### Did the same happen when running OctoPrint in safe mode?

yes but I had to cheat a little bit so there is at least one tab hidden : Using the Firefox dev tool, I shrinked the tab's div ( class="tabbable span8" ) to 500px so the "timelapse" tab is being hidden
And then the problem was the same

If I shrink even more the div, more tabs get hidden and the problem occurs on every one of them

#### Version of OctoPrint

OctoPrint 1.4.0.dev1190+gd8eb5754 

#### Operating System running OctoPrint

Raspbien 9.4 on rpi3

#### Printer model & used firmware incl. version
prusa mendel w/ marlin 1.1.8

#### Browser and version of browser, operating system running browser
Firefox 60.0 x64 on windows 10

#### Link to octoprint.log
not sure it would help : 
[octoprint.log](https://framabin.org/?1db0e9a025ff4f20#89ksCmIOlojbHrei/bB4sdprzD9sqzH0UfB+DFXwlPs=)

#### Link to contents of terminal tab or serial.log

n/a

#### Link to contents of Javascript console in the browser

in the firefox console there is only two things concerning octoprint about mixed active content being blocked : 
```
Blocage du chargement du contenu mixte actif (mixed active content) « http://prusa.home:5000/static/vendor/font-awesome-3.2.1/fonts/fontawesome-webfont.woff?v=3.2.1 »[En savoir plus]  vendor.0c31c91b2f.js:37:69967
Blocage du chargement du contenu mixte actif (mixed active content) « http://prusa.home:5000/static/vendor/font-awesome-4.7.0/fonts/fontawesome-webfont.woff2?v=4.7.0 »[En savoir plus]  vendor.0c31c91b2f.js:37:69967
```

The javascript console is not very usefull : 
```
Less has finished and no sheets were loaded.
Starting dependency resolution...
... dependency resolution done
Initial application setup done, connecting to server...
Connected to the server
Safe mode is active. Third party plugins are disabled and cannot be enabled.
Finalizing application startup
Going to bind 26 view models...
Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate since it does not exist
User gege logged in
... binding done
Application startup complete
```

#### Screenshot(s)/video(s) showing the problem:

included above

I have read the FAQ.

gege2b, 2018-05-15:
Additional info² : Tested with microsoft Edge and the problem is the same (except the glich is not visible, but the anchor in the adress bar briefly changes and revert back)

foosel, 2018-05-15:
Problem is caused by a sanity check in the hash switch function which is supposed to not allow switching to (invisible) tabs for which the permissions are lacking. Sadly that also triggers for tabs that are invisible due to not fitting into the tab bar. Need to find a workaround for that...

gege2b, 2018-05-15:
Hi Gina

What about adding some data-* attribute on the hidden tab's div, and rely on that to know wheither or not the corresponding panel can be shown ?

something like data-authorized="no|yes" or whatever ?

foosel, 2018-05-15:
Fixed I think by the above commit, at least I cannot reproduce it anymore and based on my own understanding of how it could happen it also should not be possible to happen anymore. Thanks for reporting this!

closed 2018-05-15 by foosel

gege2b, 2018-05-16:
I just updated my OP instance and I can't reproduce it too, so I guess it's definetly OK

thank YOU for this amazing software and all the work you do on it

commit 39e6c11c (2018-05-15): Fix overflown tabdrop tabs not being selectable

Closes #2643