OctoPrint/OctoPrint #2646: [Request] Brim for supports only

slikk66, 2018-05-16:
Often times I have a model I'd like to not print with a brim on it, but then there are teeny tiny support structures (<3mm round for example) that never stick unless I add brim to the entire print.  Is a "brim for support only" option possible?

ntoff, 2018-05-17:
That's a job for your slicer, OctoPrint only provides a way to utilise the cura slicer (and a super old version at that), there's no real way for OctoPrint to control the slicer's behaviour. There are 3rd party  plugins that give more control, and access to other slicers, but again they only provide a way to change settings in the slicer, not dictate how it does its thing.

tl;dr: Not something OctoPrint has any control over

slikk66, 2018-05-17:
sorry, im an idiot.. posted in the wrong hub.

closed 2018-05-17 by slikk66