OctoPrint/OctoPrint #2649: Octoprint looses USB connection with printer repeatedly

davidsielert, 2018-05-18:
#### What were you doing?

1. Uploaded via Cura
2. Started print
3. ...

#### What did you expect to happen?
Finish print
#### What happened instead?
Print stopped mid print about 2 hours in , has happened 3 times in a row now
#### Did the same happen when running OctoPrint in safe mode?
haven't tried that

#### Version of OctoPrint
OctoPrint 1.3.8 

#### Operating System running OctoPrint
running on OctoPi 0.14.0

#### Printer model & used firmware incl. version
Ender 3 th3d unified (marlin) latest

#### Browser and version of browser, operating system running browser
Chrome on windows and osx

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

Logs are too big for gist.. Placed in public dropbox folder 
https://www.dropbox.com/sh/ivpnj9x06hcpoh8/AAC7BgmYG8RPMa-7vdhxe_o9a?dl=0

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-05-18:
What I see in your `dmesg` output is that the driver ran into an error and the USB device that's your printer got reported as disconnected twice. That `dmesg` entry at 06:04 also coincides with the `SerialException` logged in `octoprint.log` (compare the timestamps), so my money would be that this is the cause of that exception. Which makes it a problem outside of OctoPrint's control - if the OS OctoPrint is running on is even losing sight of the serial connection, there's *nothing* that OctoPrint can do about it.

I can't tell you anything that isn't already listed in [this FAQ entry](https://discourse.octoprint.org/t/octoprint-randomly-loses-connection-to-the-printer-with-a-serialexception/228?u=foosel). 

Since this isn't an issue OctoPrint can solve, I'm closing this bug, but you are welcome to get in touch on the [community forum](https://discourse.octoprint.org), maybe someone there has a hint that might help you solve this.

closed 2018-05-18 by foosel

davidsielert, 2018-06-10:
@foosel Seriously alot of people are experiencing this with reccomended hardware and setups trying all the given solutions does nothing to fix it

foosel, 2018-06-12:
Then I suggest all of those people get together in a mega thread on this issue on the [community forum](https://discourse.octoprint.org) and maybe that way we can find further possible solutions (or at least figure out the common denominator).

But as I said - when the *underlying operating system* reports the machine as disconnected, there's nothing OctoPrint running on that OS can do to fix it (or prevent it). It's the same as if you'd pulled out the USB connector from the point of view of OctoPrint here.