OctoPrint/OctoPrint #2652: OctoPrint sends different g-code sequence when called by plugin to cancel a job

kennethjiang, 2018-05-18:
#### What were you doing?

I am the author of OctoPrint Anywhere plugin. I was trouble-shooting a problem that one of my users runs into.

The problem is that when a print job is cancelled via the plugin, OctoPrint will wait for `Recv: ok` for each of the g-code it sends to cancel the job:

```
Changing monitoring state from "Printing" to "Cancelling"
Send: N15 M106 S255**123
Recv: T:215.5177
Recv: ok 14
Send: N16 M17**8
Recv: ok 15
Send: N17 G0 Z1.310794 F48**50
Recv: ok 16
Send: N18 G92 E0**84
Recv: ok 17
Send: N19 G0 Z1.310794 E10 F360**97
Recv: T:232.7578
Recv: T:229.5664
Recv: T:224.7060
Recv: T:220.7146
Recv: ok 18
Send: N20 G4 S3**117
Recv: ok 19
Send: N21 G0 X0.240000 Y2.100000 F1500.000000**120
Recv: T:219.1448
Recv: T:216.0244
Recv: T:214.5706
Recv: T:216.8483
Recv: T:214.5706
Recv: T:209.5739
Recv: T:210.3830
Recv: T:212.6119
Recv: ok 20
Send: N22 G0 X5.640000 Y6.950000 Z-1.119248 F400**34
Recv: ok 21
Send: N23 G92 E0**92
Recv: ok 22
Send: N24 G0 X29.343000 Y33.740000 Z0.029766 F1440**55
Recv: T:219.1448
Recv: T:218.3623
Recv: T:216.8022
Recv: ok 23
Send: N25 M84*40
Recv: ok 24
Send: N26 M104 T0 S0*21
Recv: T:215.2005
Recv: ok 25
Recv: ok 26
Send: N27 M106 S0*82
Changing monitoring state from "Cancelling" to "Operational"
```

Whereas when the print job is directly cancelled in OctoPrint (the *cancel* button at the upper-left corner), OctoPrint will just send all these g-code rapidly without waiting for `Recv: ok`:

```
Changing monitoring state from "Printing" to "Cancelling"
Send: N13 M109 S215**118
Recv: T:193.4766
Send: N14 G4 S15**69
Send: N15 M106 S255**123
Send: N16 M17**8
Send: N17 G0 Z1.310794 F48**50
Send: N18 G92 E0**84
Send: N19 G0 Z1.310794 E10 F360**97
Send: N20 G4 S3**117
Send: N21 G0 X0.240000 Y2.100000 F1500.000000**120
Send: N22 G0 X5.640000 Y6.950000 Z-1.119248 F400**34
Send: N23 G92 E0**92
Send: N24 G0 X29.343000 Y33.740000 Z0.029766 F1440**55
Send: N25 M84*40
Send: N26 M104 T0 S0*21
Send: N27 M106 S0*82
Changing monitoring state from "Cancelling" to "Operational"
```

OctoPrint Anywhere plugin calls `self._printer.cancel_print()` to cancel the job. However, after tracing the code I found out that `kwargs={'tags': set(['plugin:anywhere', 'source:plugin'])}` was passed to `octoprint/printer/standard.py.cancel_print ` when it was called from a plugin, compared to `kwargs={'tags': set(['api:job', 'source:api'])}` when the method was called by OctoPrint itself. I suspect this subtle difference causes OctoPrint to behave differently when sending g-code.

The practical problem that this difference causes to my user is that, when her job is cancelled directly in OctoPrint (all these g-code sent without waiting), the job is cancelled properly. Whereas when the job is cancelled using the plugin, the hotend temperature is never reset to 0.

1. Start a print job
2. Cancel the print job using OctoPrint Anywhere plugin. The plugin in turn calls `self._printer.cancel_print()`.

#### What did you expect to happen?

The job is cancelled properly, which means the hotend temperature is reset to 0.

#### What happened instead?

The printhead stops. But hotend temperature is not reset to 0.

#### Did the same happen when running OctoPrint in safe mode?

This issue is about OctoPrint behaving differently when job cancel is triggered by a plugin. By definition it can not be reproduced in safe mode.

#### Version of OctoPrint

1.3.8

#### Operating System running OctoPrint

OctoPi

#### Printer model & used firmware incl. version

M3D Micro Kickstarter Edition. Firmware Version 2016-04-04-01

#### Browser and version of browser, operating system running browser

N/A.

#### Link to octoprint.log

https://drive.google.com/open?id=1I3WcTKBkVqpz2CaODHI9d2TmbR4FOtCk

#### Link to contents of terminal tab or serial.log

https://drive.google.com/open?id=1ZwZdaWasUmK1LUFyu9LTRh-9yF-fhjHc

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-05-23:
OctoPrint always counts acknowledgements from the printer and only sends out commands if it has enough "free slots" to do so. The tags that OctoPrint adds to the commands also aren't used by itself. They are merely provided for plugins to make use of (the Octolapse plugin makes heavy use of those for example, to avoid rewriting its own injected commands).

According to your logs up there, the `M104` command to set the hotend temperature to 0 is sent just fine as well:

```
2018-05-18 03:28:16,226 - Send: N26 M104 T0 S0*21
2018-05-18 03:28:17,324 - Recv: T:215.2005
2018-05-18 03:28:18,171 - Recv: ok 25
2018-05-18 03:28:18,205 - Recv: ok 26
2018-05-18 03:28:18,207 - Send: N27 M106 S0*82
2018-05-18 03:28:18,218 - Changing monitoring state from "Cancelling" to "Operational"
2018-05-18 03:28:18,223 - Recv: ok 27
2018-05-18 03:28:18,354 - Recv: wait
2018-05-18 03:28:18,358 - Recv: T:205.6979
2018-05-18 03:28:19,358 - Recv: wait
2018-05-18 03:28:19,362 - Recv: T:205.6979
2018-05-18 03:28:20,361 - Recv: wait
2018-05-18 03:28:20,365 - Recv: T:209.5212
```

For some reason the firmware seems to completely ignore it however 🤷‍♀️

To be honest, my money here would be less on a difference between cancelling from your plugin and from OctoPrint directly (since both are triggering the exact same code internally) and more on the exact timing of the attempt to cancel in both cases.

What also strikes me as odd btw is the initial connection start:

```
2018-05-18 03:25:55,521 - Connecting to: /dev/ttyACM1
2018-05-18 03:25:55,605 - Changing monitoring state from "Offline" to "Opening serial port"
2018-05-18 03:25:55,608 - Connected to: Serial<id=0x69327470, open=True>(port='/dev/ttyACM1', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2018-05-18 03:25:55,610 - Changing monitoring state from "Opening serial port" to "Connecting"
2018-05-18 03:25:55,618 - Send: N0 M110 N0*125
2018-05-18 03:25:55,631 - Recv: ok 0
2018-05-18 03:25:55,633 - Changing monitoring state from "Connecting" to "Operational"
2018-05-18 03:25:55,642 - Send: N0 M110 N0*125
2018-05-18 03:25:55,648 - Recv: ok 0
2018-05-18 03:25:55,650 - Recv: ok 0
2018-05-18 03:25:55,651 - Recv: ok 0
2018-05-18 03:25:55,662 - Recv: ok 0
```

Notice how OctoPrint sent two `M110` here (so far so good) but for some reason got 5(!) `ok` back from the printer. I'd have expected two here. That really puts things out of whack.

kennethjiang, 2018-05-23:
The user reported that she has always been able to successfully cancel print jobs directly in OctoPrint, but doing it from the plugin has never worked (she tried several times). 

How does OctoPrint determine if there are enough "spots" to send g-code? I have worked with the user to reliably reproduce this problem. From `serial.log` it seems that, when cancelled in OctoPrint, OctoPrint sent 17 g-codes before getting any `ok`. Whereas when `cancel_print()` is called by the plugin, it only allows 1 unacknowledged g-code. Example below:

Cancelled in OctoPrint:

```
Recv: T:167.6193
Recv: T:169.6782
Recv: T:174.6981
Recv: ok 10
Send: N11 G0 X0.000000 Y1.500000 Z4.110694 F1800**50
Send: N12 M18**3
Changing monitoring state from "Printing" to "Cancelling"
Send: N13 M109 S215**118
Recv: T:193.4766
Send: N14 G4 S15**69
Send: N15 M106 S255**123
Send: N16 M17**8
Send: N17 G0 Z1.310794 F48**50
Send: N18 G92 E0**84
Send: N19 G0 Z1.310794 E10 F360**97
Send: N20 G4 S3**117
Send: N21 G0 X0.240000 Y2.100000 F1500.000000**120
Send: N22 G0 X5.640000 Y6.950000 Z-1.119248 F400**34
Send: N23 G92 E0**92
Send: N24 G0 X29.343000 Y33.740000 Z0.029766 F1440**55
Send: N25 M84*40
Send: N26 M104 T0 S0*21
Send: N27 M106 S0*82
Changing monitoring state from "Cancelling" to "Operational"
Send: N0 M110*35
Recv: ok 0
```

Cancelled by OctoPrint Anywhere plugin:

```

Send: N9 G0 Z5.000000 F48**1
Recv: ok 9
Send: N10 G28**8
Recv: T:106.0913
Recv: T:109.7117
Recv: T:117.0501
Recv: T:122.4515
Recv: T:128.6792
Recv: T:133.0168
Recv: T:136.7794
Changing monitoring state from "Printing" to "Cancelling"
Send: N11 G0 X0.000000 Y1.500000 Z4.110694 F1800**50
Recv: T:148.0980
Recv: T:148.5349
Recv: T:152.8752
Recv: T:152.0129
Recv: T:158.0672
Recv: ok 10
Send: N12 M18**3
Recv: T:187.5294
Recv: ok 11
Send: N13 M109 S215**118
Recv: T:191.0805
Recv: T:191.1506
Recv: T:194.0059
Recv: ok 12
Send: N14 G4 S15**69
Recv: T:176.8798
Recv: T:179.6394
Recv: T:179.0268
Recv: T:183.0428
Recv: T:185.3700
Recv: T:188.8970
Recv: T:189.6065
Recv: T:191.8112
Recv: T:193.2413
Recv: T:193.9585
Recv: T:196.1183
Recv: T:197.6280
Recv: T:197.4389
Recv: T:199.8070
Recv: T:199.0792
Recv: T:202.7905
Recv: T:201.8807
Recv: T:203.4096
Recv: T:206.4761
Recv: T:206.5308
Recv: T:207.2175
Recv: T:208.7048
Recv: T:209.4508
Recv: T:211.5971
Recv: T:213.1537
Recv: ok 13
Send: N15 M106 S255**123
Recv: T:214.7606
Recv: T:213.9572
Recv: T:215.4713
```

foosel, 2018-05-23:
> How does OctoPrint determine if there are enough "spots" to send g-code?

It has an internal counter. Every received `ok` (or click on the Fake Acknowledge button) increases that counter. Every sent GCODE command (with or without line number and checksum) decreases it. That's all there is to it. Note that this is tracked across the life time of the whole connection to the printer, not just the print. So if the printer sends `ok`s for which no commands were sent by OctoPrint (like it can be seen in the log you posted up there), you get an increased counter and thus lots of free spots.

When you cancel, OctoPrint enqueues the full cancel script in one go. The send queue then will take care of sending that line by line. If the internal counter for some reason (spurious `ok`s) has enough free slots to allow for sending all lines from the cancel script in one go, that will happen then. If not, it won't. So for some reason, something in the printer communication of your user frees up a ton of open slots - to understand what I'd need a full `serial.log`, ideally of a communication with a cancel from OctoPrint and then one from your plugin. That might allow to point out what's happening there.

I also just noticed that this is with an M3D. As far as I understand that won't even run without a helper plugin, and I have no idea what said plugin does to the communication. So this could be a ton of things.

What I can tell you though is that there's no place in OctoPrint that actually looks at what tags are set on the cancel call and differentiates in behaviour between sources. If you call cancel via button/REST API or from a plugin doesn't matter - it's the exact same code that runs.

foosel, 2024-05-22:
@kennethjiang is this still relevant or can this be closed?

kennethjiang, 2024-05-22:
Yes we can definitely close it. Sorry for leaving it open for so long.

closed 2024-05-22 by kennethjiang