OctoPrint/OctoPrint #2654: OctoPrint keeps logging me out

BNoiZe, 2018-05-19:
#### What were you doing?

1. Logged in to OctoPrint with "Remember me" checked
2. Startet my print and checked the status from time to time
3. OctoPrint exits me session and logged me out

#### What did you expect to happen?
Staying logged in into OctoPrint

#### What happened instead?
OctoPrint logs me out

#### Did the same happen when running OctoPrint in safe mode?
Should not be applicable here

#### Version of OctoPrint
1.3.8

#### Operating System running OctoPrint
OctoPi 0.15.0

#### Printer model & used firmware incl. version
-

#### Browser and version of browser, operating system running browser
macOS 10.13.4 using Safari 11.1 

#### Link to octoprint.log
https://gist.github.com/BNoiZe/2a7bfe258e1aafc8d230a3ea3b6ac8aa

#### Link to contents of terminal tab or serial.log
Does not matter if the printer is connected or not.

#### Link to contents of Javascript console in the browser
https://gist.github.com/BNoiZe/2a7bfe258e1aafc8d230a3ea3b6ac8aa

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

ntoff, 2018-05-20:
>Should not be applicable here

The reason safe mode is required is to rule out interference by 3rd party plugins, please repeat the tests in safe mode.

BNoiZe, 2018-05-20:
@ntoff Tested it, it keeps happening in safe mode.

ntoff, 2018-05-22:
Is anything in this thread relevant to you? https://github.com/foosel/OctoPrint/issues/1792 Are you opening safari in a normal way or via any sort of web app or desktop page shortcut? Any browser extensions that might see OctoPrint's self signed https (assuming you're accessing it via https) certificate as "odd" and nuke any trace of your having been there?

cassianoleal, 2018-05-22:
I have the issue with both Safari and Firefox, either using HTTPS or plain HTTP. Seems to go away when accessing it by IP.

I'll try running it for a bit through a hosts record pointing to the IP as opposed to the avahi-discovered one to see if it makes a difference. Any other scenarios I could help with testing?

foosel, 2018-05-22:
It could maybe be caused by changing IP addresses. I recently fixed an issue there which made the "remember me" function not work properly as well. If you feel comfortable testing a development version, the fix is available on the `maintenance` branch. Otherwise wait until the first RC of 1.3.9.

cassianoleal, 2018-05-22:
I can probably try it tomorrow. Perhaps that will also solve https://github.com/foosel/OctoPrint/issues/2658?

foosel, 2018-05-23:
Very much unlikely.

BNoiZe, 2018-05-23:
@ntoff I had the problem with the self-signed certificate and with a LE certificate too. I tried disabling the browser extension, didn't solve the problem. I'm opening Safari in a normal way. #1792 doesn't seem relevant to me.

@foosel I'm using a fixed IP on OctoPrint.

foosel, 2018-05-23:
@BNoiZe does this only happen with Safari or can you reproduce it also on another browser? Just yesterday I ran into yet another Safari-exclusive bug caused by not adhering to existing standards, and I'm honestly leaning towards declaring that browser as not supported by now due to all the issues it causes. Almost feels like it's the new IE.

cassianoleal, 2018-05-23:
I don't think my IP changes either, and surely not 5 times within the space of a minute or two, which was probably the highest frequency of relogins I had to do so far. Even when not that bad, the lost sessions are quite common, happening at least a few times a day on a 24/7 connected system, which should never change IPs as its DHCP lease would be renewed frequently (and I'm almost sure I've frozen that allocation to the Pi's MAC but I might be wrong there).

About browsers, I'm using mostly Firefox but have had the same behaviour on Safari as well.

I don't do web development but I like Safari from the POV of a user, it's a good and performant browser. It saddens me to hear that it's deviating heavily from standards... :(

BNoiZe, 2018-05-23:
It also happens in Firefox and Chrome for me.

crzcrz, 2018-05-24:
I have been haunted by this issue since I installed Octoprint. I'm running it on a RPi, via OctoPi. Among other things that means that haproxy is involved.

What I've noticed so far is that "remember me"  cookie does not seem to get stored. It is easy to reproduce:
* Open Octoprint in Safari (via _octopi.local_ in my case).
* Check _Storage/Cookies_ in Web Inspector. There will be a session cookie named `session_P5000`.
* Log in and check "Remember me".
* Check _Storage/Cookies_ again, there will be no change and only the session cookie present.

It makes no difference whether Octoprint is accessed by HTTP or HTTPS (with a self-signed certificate).

The behaviour is different when running Octoprint from the source code and accessing it at _http://127.0.0.1:5000_. Immediately after logging in (and checking "Remember me"), another cookie appears, named `remember_token_P5000`. However it seems to have a wrong expiration time, happening to be exactly the time of logging in. I'd expect it to be set somewhere in the future.

I don't know if any of that is helpful or not.

anoldguy, 2018-05-25:
This happens to me on macOS 10.13.4 with Safari, FF, and Chrome, and on Ubuntu 18.04 with FF and Chrome.

All browsers are current and up to date.

foosel, 2018-05-25:
Something that I just remembered when looking up the `remember_me` cookie commit (42fb4f47008c9168b1657f9a7a9ea6958228c530) - do all of you experiencing this run IPv6 in their home LAN? I remember running into an issue there with multi-homed Win10 machines, and in the current stable version there might also still be an issue with remote address detection that could explain session invalidation.

Anyone here reliably reproducing this capable and willing of testing the `maintenance` branch?

joshwills, 2018-05-25:
Screenshots/more browser logs might help here - I'd like to know what "exits my session" looks like (what error/message/etc. is shown), and also network traffic from the browser:

2018-05-19 13:35:24,443 - octoprint.server.util.sockjs - INFO - Client connection closed: 2001:4bca:102:f794:215a:2e61:c329:a3e2
2018-05-19 13:35:24,873 -  octoprint.server.util.sockjs - INFO - New connection from client:  2001:4bca:102:f794:215a:2e61:c329:a3e2

looks weird/too rapid.  A screenshot at least of the network traffic could help, though some detailed responses for any that failed would be great too.

anoldguy, 2018-05-25:
Yes, I do run both ipv4 and ipv6. I’ll upgrade to the maintenance branch this weekend and see how it goes.

foosel, 2018-05-25:
It might also be interesting to enable the tornado access log since that will log all requests incl. their methods, paths, return codes and client ips:

[image]

Especially the latter might help identify any IP toggling situations that would cause session invalidation (on 1.3.8). This is btw something anyone can easily try without having to switch to the `maintenance` branch.

BNoiZe, 2018-06-03:
@foosel I switched to the maintenance RCs, but the problem persists. As you suggested I enabled the tornado.access logging. See https://gist.github.com/BNoiZe/6b2243ed2bb30a557282db0c5b0f95c0

shbatm, 2018-06-21:
I am also having this issue when accessing the Octoprint instance from my Windows 10 machine with Chrome.  Same scenario, regardless of if I choose remember me or not, after an unspecified amount of time I will be logged out.

On several occasions the UI will not even refresh to notify me that I have been logged out (i.e. I go to click Cancel Print and nothing happens, when I refresh the page, it shows me as logged out). There has also been occasions when I log back in and do not have access to any functions (i.e. the connection box appears, I click Connect and nothing happens; I have to refresh page again (Ctrl+Reload) to find I'm logged out again. 

I have enabled the tornado.access logs and will post back when I capture something.

BNoiZe, 2018-06-22:
I just checked this again. Safari keeped logging me out. After I upgraded to RC1 the problem dissolved into thin air.

shbatm, 2018-06-23:
Same for me. But RC1 broke my GCODE scripts... Off to fix that issue.

foosel, 2018-06-25:
> But RC1 broke my GCODE scripts

In what way?

shbatm, 2018-06-25:
My GCODE Scripts for After Pause and After Cancelled seemed not to run at all. The print would cancel, but the script tells Z to raise and the bed to push forward, but nothing happened.  Was going to dig in further and open a separate issue, but since had to leave town and haven't had a chance to follow up.
[GCODE Scripts.txt](https://github.com/foosel/OctoPrint/files/2133572/GCODE.Scripts.txt)