OctoPrint/OctoPrint #2656: Error:Line Number is not Last Line Number+1, Last Line:

KrisRux, 2018-05-20:
#### What were you doing?

OctoPrint installed via OctoPi image on a raspi zero w

    start print
    randomly, the printer just stops, in the communication log the following appears:

```
2018-05-20 18:23:24,029 - Send: N26 G1 X53.073 Y-0.330 E0.0013 F1800*20
2018-05-20 18:23:24,033 - Recv: ok
2018-05-20 18:23:24,037 - Send: N27 G1 X53.073 Y0.330 E0.0189*85
2018-05-20 18:23:24,041 - Recv: ok T:192.89 /195 B:62.15 /60 B@:0 @:85 DHT Temp:20.60, Humidity:45.10
2018-05-20 18:23:24,045 - Send: N28 G1 X53.071 Y0.380 E0.0202*83
2018-05-20 18:23:24,060 - Recv: ok T:192.89 /195 B:62.15 /60 B@:0 @:85 DHT Temp:20.60, Humidity:45.10
2018-05-20 18:23:24,075 - Send: N29 G1 X52.044 Y10.477 E0.2903*99
2018-05-20 18:23:24,077 - Recv: ok T:192.89 /195 B:62.15 /60 B@:0 @:85 DHT Temp:20.60, Humidity:45.10
2018-05-20 18:23:24,093 - Send: N30 G1 X52.036 Y10.527 E0.2916*110
2018-05-20 18:23:24,094 - Recv: ok T:192.89 /195 B:62.15 /60 B@:0 @:85 DHT Temp:20.60, Humidity:45.10
2018-05-20 18:23:24,108 - Send: N31 G1 X51.904 Y11.173 E0.3092*100
2018-05-20 18:23:24,113 - Recv: ok T:192.89 /195 B:62.15 /60 B@:0 @:85 DHT Temp:20.60, Humidity:45.10
2018-05-20 18:23:24,129 - Send: N32 G1 X51.891 Y11.222 E0.3105*98
2018-05-20 18:23:24,132 - Recv: X:53.07 Y:-0.38 Z:0.200 E:0.0000
2018-05-20 18:23:24,140 - Recv: ok
2018-05-20 18:23:24,144 - Recv: ok
2018-05-20 18:23:24,149 - Send: N33 G1 X48.853 Y20.906 E0.5806*102
2018-05-20 18:23:24,150 - Recv: Error:Line Number is not Last Line Number+1, Last Line: 26
2018-05-20 18:23:24,152 - Send: N34 G1 X48.835 Y20.954 E0.5819*104
2018-05-20 18:23:24,154 - Recv: Resend:27
2018-05-20 18:23:24,160 - Send: N27 G1 X53.073 Y0.330 E0.0189*85
2018-05-20 18:23:24,161 - Recv: ok
2018-05-20 18:23:24,163 - Send: N28 G1 X53.071 Y0.380 E0.0202*83
2018-05-20 18:23:24,164 - Recv: Echo:Unknown command: "dity:45.53.073 Y0.330 E0.0189"
2018-05-20 18:23:24,167 - Recv: ok
2018-05-20 18:23:24,169 - Send: N29 G1 X52.044 Y10.477 E0.2903*99
2018-05-20 18:23:24,170 - Recv: Error:Line Number is not Last Line Number+1, Last Line: 26
2018-05-20 18:23:24,172 - Recv: Resend:27
2018-05-20 18:23:24,176 - Recv: ok
``

#### Version of OctoPrint

OctoPrint 1.3.8 running on OctoPi 0.14.0

#### Operating System running OctoPrint

Raspberry

#### Printer model & used firmware incl. version

Kossel XL, firmware Mk4duo 3.4.6

[serial.log](https://github.com/foosel/OctoPrint/files/2020555/serial.log)

KrisRux, 2018-05-20:
[serial.log](https://github.com/foosel/OctoPrint/files/2020562/serial.log)

foosel, 2018-05-22:
Already replied to you [here](https://discourse.octoprint.org/t/error-line-number-is-not-last-line-number-1-last-line/1934/3). Most likely caused by a combination of this extraordinarily long `M105` response which overwhelms the serial buffer (and thus a duplicate of #2454) and a possibly noisy line.

closed 2018-05-22 by foosel