OctoPrint/OctoPrint #2657: Format errors after update to 1.3.8

JonasGe, 2018-05-21:
#### What were you doing?

1. Updated OctoPrint to the latest version
2. Restarted system
3. Uploaded GCode and started print

#### What did you expect to happen?
The print to start like any other print before

#### What happened instead?

After heating, the Z calibration started like normal and then the printer stopped doing anything, receiving format errors.

`Recv: TargetBed:60
Recv: T:104.54 /0 B:53.05 /60 B@:255 @:0
SNIPPED
Recv: T:90.34 /0 B:59.46 /60 B@:244 @:0
Recv: ok 3
Send: N4 M105*35
Recv: TargetExtr0:205
Recv: ok 4
Send: N5 M117 [          ]   0%*50
Recv: T:90.34 /205 B:59.52 /60 B@:174 @:0
Recv: ok 5
Send: N6 M109 S205*105
Recv: ok 6
Send: N7 M82*30
Recv: TargetExtr0:205
Recv: T:89.66 /205 B:59.70 /60 B@:197 @:255
SNIPPED
Recv: T:135.07 /205 B:60.41 /60 B@:0 @:255
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N8 M105*47
Recv: T:136.99 /205 B:60.53 /60 B@:0 @:255
SNIPPED
Recv: T:189.90 /205 B:60.37 /60 B@:7 @:255
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N9 M105*46
Recv: T:191.41 /205 B:60.37 /60 B@:7 @:255
SNIPPED
Recv: T:204.35 /205 B:60.37 /60 B@:86 @:155
Recv: ok 7
Send: N10 M105*22
Recv: ok 8
Send: N11 G21*42
Recv: T:204.93 /205 B:60.37 /60 B@:7 @:25
Recv: ok 9
Send: N12 G90*35
Recv: T:204.93 /205 B:60.37 /60 B@:7 @:25
Recv: ok 10
Send: N13 M107*23
Recv: T:204.93 /205 B:60.37 /60 B@:7 @:25
Recv: ok 11
Send: N14 G28*38
Recv: ok 12
Send: N15 G92 E0*115
Recv: ok 13
Send: N16 G1 F200 E3*29
Recv: ok 14
Send: N17 G92 E0*113
Recv: ok 15
Send: N18 G1 F{travel_speed}*81
Recv: X:0.00 Y:0.00 Z:313.80 E:0.00
Recv: ok 16
Send: N19 M117 Printing...*35
Recv: ok 17
Send: N20 M107*23
Recv: Error:Format error
Recv: 
Recv: Resend:18
Send: N18 G1 F{travel_speed}*81
Recv: ok
Send: N19 M117 Printing...*35
Recv: skip 19
Recv: ok
Send: N20 M107*23
Recv: skip 20
Recv: ok
Send: N21 G0 F3600 X29.497 Y18.591 Z0.3*42
Recv: Error:Format error
Recv: 
Recv: Resend:18
Send: N18 G1 F{travel_speed}*81
Recv: ok
Send: N19 M117 Printing...*35
Recv: skip 19
Recv: ok
Send: N20 M107*23
Recv: skip 20
Recv: ok
Send: N21 G0 F3600 X29.497 Y18.591 Z0.3*42
Recv: skip 21
Recv: ok
Send: N22 G1 F1800 X29.341 Y18.838 E0.01603*62
Recv: Error:Format error`

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint
1.3.8

#### Operating System running OctoPrint
OctoPi

#### Printer model & used firmware incl. version
Kossel Mini with Marlin

#### Browser and version of browser, operating system running browser
Chrome on W10

#### Link to octoprint.log
https://gist.github.com/JonasGe/859b2ac1043783809eb58e4d90016218

#### Link to contents of terminal tab or serial.log
See above

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ. Y

JonasGe, 2018-05-21:
Looking back at the log, I can see a G1 F{travel_speed}. I assume this should be filled in by the slicer?

JonasGe, 2018-05-21:
Removed the travel_speed and now it works again.

I'll close this issue myself but just for the record: I think this is a (breaking?) change from 1.3.6 to 1.3.8 as I never experienced this issue before and never touched the start-script of my slicer.

closed 2018-05-21 by JonasGe

foosel, 2018-05-22:
The error comes from your firmware, not OctoPrint. If you were sending these lines with placeholders in the past and your firmware didn't produce an error then, it sounds more like your firmware's behaviour changed. OctoPrint 1.3.6 and 1.3.8 will both just happily send everything from a streamed file to your printer. No special voodoo takes place there that would have prevented this error in 1.3.6 but make it surface in 1.3.8.

MTKnife, 2019-01-10:
I'm having the same issue on a Maker Select v2, and I'm quite sure there wasn't a printer firmware update, since updating the firmware requires opening the case, disattaching the LED screen, and flashing the thing with an Arduino programmer.  And not to put too fine a point on it, but it happened _precisely_ when Octoprint updated:  I did multiple prints that day, and got no errors on the earlier ones, and errors on every print since.  Something changed due to the update of Octoprint, to the point where it's not quite unusable, but very unpleasant, since I have to keep hitting "Fake Acknowledgement" over and over, or, as described above, edit the gcode.

brazda, 2019-01-20:
I seem to be getting the same issue however if I power cycle my MP select v2 and home and preheat the PLA from the nob then run the print from OctoPrint it seems to work. Next print I am going to just home it and see if it works after this print.

joshbillions, 2019-02-13:
I'm having the same issue on my Maker Select v2. Definitely haven't updated the firmware.

nitedmn, 2019-03-29:
Was there ever a solution to this? I just setup OctoPrint, with my MP Maker Select V2 and I'm having the same problem as brazda. If I pre-heat and home all, from the printer, then start the print through OctoPrint, it works fine. The same gcode file, run from the SD card in the printer, will also work fine.

brazda, 2019-03-29:
Well heck I don't remember how I fixed mine lol. I changed a lot since this post. I removed all my start gcode and added back only stuff I wanted one by one. I can't remember if this was before or after I changed printer firmware too. Also I quit using auto detection for baud rate don't remember why.
I am in the process now to move to klipper on all my printers as it can use the PI and octoprint to run at 32 bit rather than 8 bit.

nitedmn, 2019-03-29:
Looks like I figured it out. 

For anyone else who has this problem, and stumbles on this page (it was one of the first results that came  up). You need to change start and end G-Code in Cura. Simply change all instance of {travel_speed} to {speed_travel} and everything seems to work correctly.

foosel, 2019-03-29:
I've also just added [an FAQ entry on this](https://community.octoprint.org/t/when-i-try-to-print-something-i-keep-getting-format-errors-from-the-printer/8640)...

Locking this now.