OctoPrint/OctoPrint #2661: Printer requested line 634 but no sufficient history is available, can't resend

nikoli280397, 2018-05-23:
Hi i get this error when connecting to my Da vinci 1.0 with repetier setup.

i also attach any logs
[octoprint (1).log](https://github.com/foosel/OctoPrint/files/2030815/octoprint.1.log)

Printer requested line 634 but no sufficient history is available, can't resend
Recv: ok
Recv: 
Recv: Resend:634
Printer requested line 634 but no sufficient history is available, can't resend
Recv: ok
Recv: 
Recv: Resend:634
Printer requested line 634 but no sufficient history is available, can't resend
Recv: ok
Recv: 
Recv: Resend:634
Printer requested line 634 but no sufficient history is available, can't resend
Recv: ok
Recv: 
Recv: Resend:634
Printer requested line 634 but no sufficient history is available, can't resend
Recv: ok
Recv: 
Recv: Resend:634
Printer requested line 634 but no sufficient history is available, can't resend
Recv: ok
Recv: 
Recv: Resend:634
Printer requested line 634 but no sufficient history is available, can't resend
Recv: ok
Recv: 
Recv: Resend:634
Printer requested line 634 but no sufficient history is available, can't resend
Recv: ok
Recv: 
Recv: Resend:634
Printer requested line 634 but no sufficient history is available, can't resend
Recv: ok
Recv: 
Recv: Resend:634
Printer requested line 634 but no sufficient history is available, can't resend

nikoli280397, 2018-05-23:
i use Version 1.3.8

nikoli280397, 2018-05-23:
I think this has the information needed and i think its a bug

foosel, 2018-05-23:
You are asked to follow the template for a reason.

That being said, a quick look into your log reveals that your printer asks for a resend request *right after connect*:

```
| Changing monitoring state from "Offline" to "Detecting serial port"
| Serial port list: ['/dev/ttyACM0']
| Connecting to: /dev/ttyACM0
| Changing monitoring state from "Detecting serial port" to "Opening serial port"
| Connected to: Serial<id=0x76253ab0, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
| Starting baud rate detection
| Changing monitoring state from "Opening serial port" to "Detecting baudrate"
| Trying baudrate: 115200
| Recv: Error:Checksum required when switching back to ASCII protocol.
| Recv: 
| Baudrate test retry: 4
| Recv: Resend:634
| Send: N0 M110 N0*125
| Printer requested line 634 but no sufficient history is available, can't resend
```

This may well be a bug, but it isn't a bug in OctoPrint but rather the behaviour of your printer's firmware. OctoPrint tries to reset the line number to 0 with the `M110`, your printer or rather its firmware simply ignores this though and keeps on screaming for a line OctoPrint never sent (since it wasn't even connected before being bombarded with resend requests).

A printer that would reset on connect wouldn't have this problem. A printer that would accept a line number reset while in this error condition would not have this problem. And from the looks of it this is caused by your printer a) not resetting and b) also being switched to the repetier binary mode and that's what's got it utterly confused.

This needs to be solved in your printer, not OctoPrint.

closed 2018-05-23 by foosel