OctoPrint/OctoPrint #2662: Can't set temperature with UI controls

gege2b, 2018-05-23:
Hi
#### What were you doing?

I wanted to waste some energy so I typed "70" in the heatbed control on the UI, validating with "Enter", and nothing
Tried with hotend control too, same problem, temperature command is not sent.
Tried with the +/- buttons and still no reaction

Predefined temperature dropdown is working, and the heatbed/hotend begin to drink eletrons

It's only an UI problem, temperature command sent by GCode files are OK

#### What did you expect to happen?
Electron to flow and warm my heatbed/hotend 

#### What happened instead?
Nothing

#### Did the same happen when running OctoPrint in safe mode?

yes

#### Version of OctoPrint

OctoPrint 1.4.0.dev1253+g7b773f6f but it occurs since a few updates, I just manager to fill this request now

#### Operating System running OctoPrint

raspbian 9.4 on rpi3

#### Printer model & used firmware incl. version

prusa mendel with marlin 1.1.8

#### Browser and version of browser, operating system running browser

firefox 60 on win10

#### Link to octoprint.log

[log](https://framabin.org/?a89b6fea1fde49a5#iBvbRYb/v/bbxj6TeIZi9s/4XA9hrrQ3pKxLOkmrpkg=)

#### Link to contents of terminal tab or serial.log
Forgot it
And as I'm printing right now, I will post the terminal contents when finished if it's needed

#### Link to contents of Javascript console in the browser
there is nothing except some lines saying that progress bar and search form was resized

I have read the FAQ.

ntoff, 2018-05-23:
Same result here on chrome
>Version 65.0.3325.181 (Official Build) Built on Ubuntu , running on LinuxMint 18.2 (64-bit)

js console

[image]

Nothing in the terminal either besides the regular M105 temperature query.

The temperature fine controls (+ and - buttons) do send temperature adjustments, if you type in a temperature, say 241 and then press the - button to get 240, it will correctly send the temperature request.

foosel, 2018-05-24:
Probably another case of an error with the new permission system, wrongly named permission or somesuch

foosel, 2018-05-24:
Yep, that was it, slightly different variant though. Fixed.

commit a92372ec (2018-05-24): Fix permission on temperature set button

Closes #2662
closed 2018-05-24 by foosel