OctoPrint/OctoPrint #2663: Can't set a a float feedrate through the API, only int

mamoit, 2018-05-23:
#### What were you doing?

Trying to change the feedrate with the API by post to `/api/printer/printhead`

#### What did you expect to happen?
For the feedrate change to accept a float as factor as specified [here](http://docs.octoprint.org/en/master/api/printer.html#post--api-printer-printhead).

#### What happened instead?
It returned a `400`.

#### Did the same happen when running OctoPrint in safe mode?
I did not test it in safe mode, but since the endpoint is working with integer factors, it shouldn't matter.

#### Version of OctoPrint
1.3.8

#### Operating System running OctoPrint
Ubuntu amd64

#### Printer model & used firmware incl. version
Blocks Zero (unreleased firmware, based on Marlin 1.1.8)

#### Browser and version of browser, operating system running browser
Not applicable

#### Link to octoprint.log
```
2018-05-23 17:19:07,897 - tornado.access - WARNING - 400 POST /api/printer/printhead (127.0.0.1) 5.25ms
```

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser
Not using a browser

#### Screenshot(s)/video(s) showing the problem:
No ui interaction

I have read the FAQ.

foosel, 2018-05-23:
`float`s should be accepted just fine. However there's a range limit of 50 to 200 in place. What value did you try to set?

mamoit, 2018-05-23:
Exactly between 50 and 200.
Mostly around 150.

foosel, 2018-05-25:
Ah... That seems to be more of a missing example in the documentation then. Note this (emphasis mine):

> `factor`: The new factor, percentage as integer or **float (percentage divided by 100)** between 50 and 200%.

What this means is, for 125% send either `{"command": "feedrate", "factor": 125}` or `{"command": "feedrate", "factor": 1.25}`. 

I'll add a corresponding example to make this more clear.

foosel, 2018-05-25:
This has now been clarified in the docs as already visible [here](http://docs.octoprint.org/en/maintenance/api/printer.html).

mamoit, 2018-05-25:
Confirmed!
Thanks for the clarification.

commit ce5275c9 (2018-05-25): docs: Add example for float factors

Closes #2663
closed 2018-05-25 by mamoit