OctoPrint/OctoPrint #2667: Smoothieware integration - issues

IdefixRC, 2018-05-29:
Trying to get my Tevo Little Monster on Smoothieware to work with Octoprint (latest stable version) on a PI 3 B+. Followed the instructions here: https://github.com/foosel/OctoPrint/wiki/Setup-OctoPrint-with-Smoothie 3

Implemented the Safe Method.
Initially, it failed as sudo anything required a password.
Solved this by adding mount, umount, sync to visudo to disable the password prompt for these.
(maybe this is something that should be added to the Wiki?)

The script works fine now and copies (or in my case moved) files from the PI SD to the Smoothieboard SD Card. BUT the files do not show up in the Octoprint file manager (moved file disappears) till I either reboot the PI or manually run the M20 command in the Gcode window.

Hitting the refresh button in the file manager does nothing either.

I assume that the refresh command bound to the refresh button is wrong for the smoothieboard as a manual M20 refreshes it fine. As I could not find a config option for that I'm raising this as a bug here.

Thanks !

IdefixRC, 2018-05-30:
#### What were you doing?
Ok, here we go again. :-)
Trying to get my Tevo Little Monster on Smoothieware to work with Octoprint on a PI 3 B+. Followed the instructions here: https://github.com/foosel/OctoPrint/wiki/Setup-OctoPrint-with-Smoothie 3

Implemented the Safe Method.
Initially, it failed as sudo anything required a password.
Solved this by adding mount, umount, sync to visudo to disable the password prompt for these.
(I have in the meantime added this to the Wiki page above)

The script works fine now and copies (or in my case moved) files from the PI SD to the Smoothieboard SD Card. BUT the files do not show up in the Octoprint file manager after they are moved to the smoothieware SD card (the moved file disappears from the internal SD card after the move is complete so at least that refresh works apparently) till I either reboot the PI or manually run the M20 command in the Gcode window.

Hitting the refresh button in the file manager does nothing either.

I assume that the refresh command bound to the refresh button is wrong for the smoothieboard (only refresh OctoPi SD) as a manual M20 refreshes it fine. As I could not find a config option for that I'm raising this as a bug here.

Thanks !

#### What did you expect to happen?
Refresh button to refresh SD Card content

#### What happened instead?
Did not refresh SD card content

#### Did the same happen when running OctoPrint in safe mode?
Not tested

#### Version of OctoPrint
OctoPrint 1.3.8 running on OctoPi 0.15.0

#### Operating System running OctoPrint
OctoPi

#### Printer model & used firmware incl. version
Tevo Little Monster - Smoothieware December 2017 release (latest stable)

#### Browser and version of browser, operating system running browser
Win10, Chrome Browser

#### Link to octoprint.log
N/A

#### Link to contents of terminal tab or serial.log
N/A

#### Link to contents of Javascript console in the browser
N/A

#### Screenshot(s)/video(s) showing the problem:
N/A

I have read the FAQ.

foosel, 2018-05-30:
> Link to octoprint.log
> N/A
>
>Link to contents of terminal tab or serial.log
>N/A

Please provide _all_ requested information. Without that information this ticket cannot be looked into and will be closed. Thanks.

closed 2018-06-14 by foosel