OctoPrint/OctoPrint #2668: [Request] M117 utf8 support

ak1976khv, 2018-05-30:
When I send a command M117 with Cyrillic characters through the terminal, I get a string of characters "???????" on the printer screen.
Marlin firmware supports Cyrillic characters in the command M117.
utf-8 support in the command M117 is needed to localize the "Detailed Progress" plug-in.

When I change the file /src/octoprint/util/comm.to the line
command_to_send = command.encode ("ascii", errors = "replace")
to
command_to_send = command.encode ("utf-8", errors = "replace")
the support of the cyrillic characters in the command M117 begins to work

But the file must be changed with each update of OctoPrint.

Please, add utf-8 support to the M117 command

foosel, 2018-05-30:
Hm, I don't feel comfortable with this. I can't just blindly add support for UTF-8 on sent commands since this might seriously break printers out there whose firmware does only support ASCII (or start interpreting single parts of UTF-8 characters as control chars). And at least the responses from the printer are expected to be ASCII only [according to the only thing that we have that resembles a spec](https://reprap.org/wiki/G-code#Replies_from_the_RepRap_machine_to_the_host_computer):

> **Replies from the RepRap machine to the host computer**
>
> All communication is in printable ASCII characters. [...]

So it's fair to assume that a lot of printers out there running one of the thousands of firmware variants would hiccup on UTF-8 hard.

We'd first need to find consensus on expected encoding on the interface in the 3d printing community, and I don't see this happening with the fragmentation that's going on.

ak1976khv, 2018-05-30:
Maybe, can be implemented configuration option.
Users whose printers support utf-8 will be able to add its support to Octoprint.

alexws54tk, 2018-08-22:
@ak1976khv at now in marlin is three ways to realising of russian language on screen of the printer:
* western encoding;
* japanise encoding;
* cyrillic encoding.
the choice of option of display depends on origin of the screen of the printer.
Either you __Learn English__ or you __write translit__.

@foosel may close this issue.

ak1976khv, 2018-08-23:
The Marlin firmware supports russian language on my printer's screen, but OctoPrint don't support utf-8 in M177 command.
The Problem and workaround described [here](http://3dtoday.ru/blogs/ak1976/octoprint-russkie-bukvy-v-m117/)

cp2004, 2022-02-17:
Implemented with #4390