OctoPrint/OctoPrint #2669: [Request] Include finished prints in file search results

AcHub, 2018-05-30:
I usually hide successfull prints from the file list and I appreciate that Octoprint has this option, as it makes my file list nice and clean. 
But when I search for a certain print via the file search I would like the search results to include successful prints as well, as in my case the list of failed or outstanding prints is typically rather short, and I actually only ever use the search bar when I want to find an old successful print that I want to reprint. 
So right now I have to deactivate the "hide successful prints from list" option, search the file and then reactivate the option again, which seems like an unnecesary pain. 

I guess a feature to include hidden prints in the search results could be just another option besides to "hide successful". Also search results could list matching results from the "fresh" print queue first and then append the matching results from the hidden list, so there will still be a clear separation between "fresh" and "old" prints

foosel, 2019-05-03:
Implemented by #3126, to be rolled out as part of 1.3.12.

foosel, 2019-10-22:
1.3.12 has been released.

closed 2019-10-22 by foosel