OctoPrint/OctoPrint #2673: [Request] Integrate filament runout detection

kortschak, 2018-06-03:
A number of printers now are providing filament runout detection. This appears not to be handled will (at all?) by OctoPrint. It would be nice to be able have the convenience and features of printing via OctoPrint with the security of knowing that a filament runout or breakage will not cause print failure that would otherwise have been easily recoverable.

foosel, 2018-06-04:
If the printer firmwares in question [implement the action commands that OctoPrint has been supporting for several years now](http://docs.octoprint.org/en/master/features/action_commands.html) (and which also [just have been extended by a couple new ones on the `maintenance` branch](http://docs.octoprint.org/en/maintenance/features/action_commands.html) this is already possibly just fine. Nothing to do from OctoPrint's side here really, it's the firmware support that's still lacking.

closed 2018-06-04 by foosel

kortschak, 2018-06-04:
Thanks, I'll raise it there.