OctoPrint/OctoPrint #2674: IOError: [Errno 13] Permission denied

snowdd1, 2018-06-04:
#### What were you doing?

1. Uploading gcode file
2. start print
3. watch it fail after 40 min in the print  

Logfile says "raise an issue" here

#### What did you expect to happen?  
send gcode to printer

#### What happened instead?  
connection lost, error message in the log

#### Did the same happen when running OctoPrint in safe mode?
yes

#### Version of OctoPrint
1.3.8

#### Operating System running OctoPrint
Raspbian/OctoPi 0.13.0 with manual updates using apt-get update/upgrade

#### Printer model & used firmware incl. version
Anet A8 v1, stock firmware

#### Browser and version of browser, operating system running browser
Firefox 60.0.1 64bit, Win10pro 1804

#### Link to octoprint.log
https://gist.github.com/snowdd1/0af723da93fb35178e8bcb230f76a695

#### Link to contents of terminal tab or serial.log
Though it's very blank:   
https://gist.github.com/snowdd1/90f32632b4e174ee816f4efaf28b72e4 

#### Link to contents of Javascript console in the browser
n/a

#### Screenshot(s)/video(s) showing the problem:
n/a

I have read the FAQ.

foosel, 2018-06-05:
Looks like OctoPrint lost read access to the file it was printing. That permission denied got triggered in the (python core) code for reading from the file. Not much I can do about that. Considering that you are running a fairly old version of OctoPi I assume the SD card things are running on is fairly old as well. Maybe it's starting to fail and causing this problem.

It definitely has nothing to do with PLA vs TPU as mentioned in the comment on your log, OctoPrint doesn't know that. It doesn't even really look at the lines it's sending, it just strips comments and pumps stuff to your printer. But it can't of course when the source it's reading from suddenly stops being readable.

I'll look into catching these kinds of errors and providing a more meaningful message (something like "can't read from file" or something like that), but other than that I don't think I can do anything here.

snowdd1, 2018-06-05:
Thanks for looking into this so fast! Actually, the raspi is booting from an NFS share, not the SD-card. But I'll try exactly that, given the SD Card is almost new - and the maybe the network connection is flaky. Will try switching between LAN and WLAN, too.

snowdd1, 2018-06-05:
Interesting might be that I had several good prints (never had this connection break before) with the same configuration (same Octoprint, same update level of the PI) which took even over 24h in the last weeks.

Thanks for this great program!

snowdd1, 2018-06-05:
Not really a test, but an attempt: As it says something about CODECS in the error message, I disconnected the raspicam from the unit. And as the only difference to me between the PLA and the TPU prints were much more heat to the hotend (and thus on the controller board for it, above which the raspi is located in the encasing), I mounted a small 12V fan to the encasing to help air flow (with those unusual high temperatures here in Germany this week, convection did not seem sufficient). The raspi might have gotten a little to hot for its network interface, though I couldn't find anything in the sys logs.
 
The print with the additional fan is good so far, much further advanced than I managed to get it yesterday.

I hereby close the original issue verbally, but leave this technically open for you to remember to work on the error message.

Thank you so much.

foosel, 2018-09-24:
There should now better error handling per the above commit in case a file becomes unreadable. Sadly it was impossible for me so far to trigger this error condition intentionally and on demand, so I haven't been able to properly test it. If push comes to shove though, the worst it can do is still crash the serial loop and log a corresponding message to the log, so nothing lost.

Available on `maintenance`, soon `devel`, to be released with 1.3.10.

foosel, 2018-12-10:
1.3.10 has been released.

closed 2018-12-10 by foosel