OctoPrint/OctoPrint #2675: ZeroDivisionError: float division by zero

rosterloh, 2018-06-06:
#### What were you doing?

Loading a print from the SD card which was sliced for the Prusa MK2S with Simplify3D

#### What did you expect to happen?
Octoprint to continue functioning

#### What happened instead?

Serial connection the the printer disconnected

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint
OctoPrint 1.3.8

#### Operating System running OctoPrint
OctoPi 0.15.1

#### Printer model & used firmware incl. version
Prusa i3 MK2S firmware 3.1.0

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

https://gist.github.com/rosterloh/d62c612de0239ab92a79a6e908961b24

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-06-06:
You description of what you were doing when this happened (printing from SD) doesn't match what I see in the log (sending a file to SD). So which one was it?

foosel, 2018-06-06:
Oh, and as a side note:

> **Printer model & used firmware incl. version**
> Prusa i3 MK2S firmware 3.1.0

[That firmware has serious communication issues and should therefore not be used with any kind of host program (incl. OctoPrint)](https://discourse.octoprint.org/t/printer-specific-original-prusa/239#headline--communication-issues-3.1.0).

rosterloh, 2018-06-07:
It seems to be working now. The error just asked me to submit it so trying to be a good citizen. Thanks for the tip about the firmware.

foosel, 2018-06-09:
And it was good that you reported it. I'm still not sure how that got triggered in the first place (based on what I saw in the code that could only ever possibly caused by trying to stream a file to the printer that's either empty or only consists of comments that will be stripped), but it shouldn't be possible to trigger it in the future.

Fix is pushed to `maintenance`, soon `devel`, to be released with 1.3.9

commit 534f3aa2 (2018-06-09): Fix for potential division by zero on SD upload

Closes #2675
closed 2018-07-25 by foosel