OctoPrint/OctoPrint #2676: [Request] Optimized webinterface for official Raspberry Pi 7" Touchscreen

mrcg, 2018-06-06:
As raspberry CPU are already powerfull enough to handle octoprint with touchscreen control i would like to ask for optimized webinterface especially for usage with official Raspberry Pi 7" Touchscreen (https://www.raspberrypi.org/products/raspberry-pi-touch-display/).

The optimized webinterface should be accessible with special url (e.g. /index_touchscreen_7.html).
This url could be used to startup the optimized webinterace with Chrome (or Midori) in kiosk mode (similar as shown here: https://www.youtube.com/watch?v=ZYjPcC37fNI&t=9s).

Beside the 7" touchscreen it would be also great to support the widely available and cheap 3.5" touchscreen.
Anyhow ... supporting the official 7" touchscreen should have the highest priority. ;-)

Would be great if this could be implemented soon.
Thanks
Christian

ntoff, 2018-06-06:
https://plugins.octoprint.org/plugins/touchui/

mrcg, 2018-06-07:
@ntoff: I was already aware about touchui and Octoprint-TFT. Anyhow i am not satisfied with the ergonometry as it is not really optimized for specific screens.

I think it's time to develop some native solution - out of the box - which provides best look and feel. Something similar as done from Repetier Server.

foosel, 2018-06-07:
Then roll up your sleeves and [get to it](http://docs.octoprint.org/en/master/plugins/mixins.html#uiplugin) ;)

mrcg, 2018-06-07:
What a nice and useful answer.
Please close this request.

foosel, 2018-06-07:
Oh come on. You said "I think it's time to develop some native solution" - I'm all with you on that. But I'm already running way over capacity and can't do everything on my own. So if you want to see it happen, the best way would be to help making it happen, and that's what I was pointing out here.

Everyone opening feature requests here has their own priorities, and it's impossible to cater to all of them. The only way to get more done more quickly is if people start contributing more. There's a plugin system for that very reason, making it extremely easy to scratch your own itch and share what comes out of that with the community. But most people (even those who can code just fine) don't do that and instead wait for someone else to do the leg work. And considering that - as I said - I'm now at a point where I'm well over capacity, this is simply no longer going to work if people want to see their requests actually turned into reality within a sensible time frame.

LazeMSS, 2021-03-31:
There are several plugins to choose from for this feature:
https://github.com/Z-Bolt/OctoScreen
https://plugins.octoprint.org/plugins/touchui/

closed 2021-03-31 by LazeMSS

jneilliii, 2021-03-31:
Don't forget OctoDash...

https://github.com/UnchartedBull/OctoDash