OctoPrint/OctoPrint #2681: Not retrieving Announcements or Plugin list from behind proxy

StephS, 2018-06-13:
#### What were you doing?
Using Octoprint behind a proxy

1. Setup octoprint in an environment behind a corporate proxy
2. Goto plugins page
3. plugins will not load due to connection timeout

#### What did you expect to happen?
Octoprint should use settings or environment variables for proxy

#### What happened instead?
Could not get announcements or pluging

#### Did the same happen when running OctoPrint in safe mode?
yes

#### Version of OctoPrint
1.3.8

#### Operating System running OctoPrint
Octopi 0.15.1

#### Printer model & used firmware incl. version
N/A

#### Browser and version of browser, operating system running browser
Win10 Chrome

#### Link to octoprint.log
truncated log: https://gist.github.com/StephS/60b09a77dd136291c5cc1463994f8fea

#### Link to contents of terminal tab or serial.log
N/A

#### Link to contents of Javascript console in the browser
N/A

#### Screenshot(s)/video(s) showing the problem:
N/A

I have read the FAQ.

I am behind a corporate proxy, with all of the appropriate proxy environment settings. pip works along with other tools, however octoprint doesn't work. Through some digging in the code, i found that proxy support isn't being utilized in the requests library. 
By editing 
"/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements/__init__.py" line 355, in _get_channel_data_from_network, 
 "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 739, in _fetch_repository_from_url
 "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 813, in _fetch_notices_from_url\
i was able to add proxy support.

Proxy support was added by editing the requests line: r = requests.get(url, timeout=30)
The lines were replaced with: 
```
prox = {
'http': 'http://proxy.com:8080',
'https': 'http://proxy.com:8081',
}
r = requests.get(url, timeout=30, proxies=prox)
```

This is obviously a quick fix, but ideally this should be modified to pull the proxy information from either the environment, or configuration files.

foosel, 2018-06-14:
Have you tried setting the environment variables `HTTP_PROXY` and `HTTPS_PROXY` as described [here](http://docs.python-requests.org/en/master/user/advanced/#proxies)? As far as I understand requests should already honor that without any changes in OctoPrint being necessary.

StephS, 2018-06-14:
yep. I setup my /etc/environment with HTTP_PROXY and HTTPS_PROXY
env shows them set. I also set sudoers env_keep to keep them too.
Should these be set elsewhere than /etc/environment?

Erutan409, 2018-06-14:
FWIW - I just recently setup a raspberry pi strictly as a proxy server running NGINX.  It'll forward my requests to my OctoPrint server (which is also running on a raspberry pi), without issue.  I see the announcements/updates like I always did.  My proxy configuration is also encrypting traffic via SSL (Let's Encrypt) on the public side and leaving the connection between OctoPrint and proxy server as default unencrypted communication.  I'm running 1.3.6, though.

StephS, 2018-06-15:
Thanks Gina, I did as above and it works. I did some searching and found this: https://github.com/aarnt/octopi/issues/158 

I created the profile.d script and put the exports in there, and it worked. So not really a bug, just missing documentation from OctoPi on how to properly setup the proxy.

For future reference the process is:
Create a script /etc/profile.d/proxy.sh
add the proxy lines I.E. export HTTP_PROXY=http://myproxy:8080

This will make the proxy setting global, so will work for all other services.

closed 2018-06-15 by StephS

foosel, 2018-06-18:
It's funny that you found the solution in a completely unrelated "octopi" project 😄

Thanks for the follow-up!

StephS, 2018-06-19:
so apparently that wasn't enough. it seems that it was using the cached files, so when i tried to actually install a plugin it would fail.

I ended up using an override on the service. using systemctl like this:
sudo systemctl edit octoprint
it will create override files in /etc/systemd/system/ and open an editor.
I added the lines: 
[Service]
EnvironmentFile=/etc/environment

this forces the service to run with the environment specified there.

all the other things i tried didn't work, .bashrc, /etc/profile.d/, /etc/environment
a total pain. Perhaps it would be better to add a config option as a secondary option, using the same method as the hack i used above.