OctoPrint/OctoPrint #2687: PIP Handling error

Valicek1, 2018-06-20:
#### What were you doing?
I am trying to install plugin. My octoprint is bare install under it's own user (octoprint), installed as described in the readme) under latest debian stable in virtualenv. 
1. Install system
2. Clone Octoprint
3. Install venv
4. Start octoprint --serve
5. Do the base config
6. Try to install plugin

#### What did you expect to happen?

Plugin to install as usual. Or to remove it. 

#### What happened instead?

Unknown error was thrown in web interface, this stacktrace was thrown into octoprint.log: 

```
2018-06-20 10:46:13,595 - octoprint.plugins.pluginmanager - ERROR - Could not uninstall plugin via pip
Traceback (most recent call last):
  File "/home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 495, in command_uninstall
    self._call_pip(pip_args)
  File "/home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 652, in _call_pip
    return self._pip_caller.execute(*args)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/pip.py", line 172, in execute
    return self.call(command, **kwargs)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/commandline.py", line 83, in call
    p = sarge.run(command, **kwargs)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/sarge/__init__.py", line 1462, in run
    p.run(input=input, async_=async_)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/sarge/__init__.py", line 1071, in run
    self.run_node(node, input=input, async_=False)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/sarge/__init__.py", line 1187, in run_node
    result = getattr(self, method)(node, input, async_)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/sarge/__init__.py", line 1333, in run_command_node
    node.cmd.run(input=input, async_=async_)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/sarge/__init__.py", line 655, in run
    self.process = p = Popen(self.args, **self.kwargs)
TypeError: __init__() got an unexpected keyword argument 'async'
2018-06-20 10:46:13,601 - tornado.access - ERROR - 500 POST /api/plugin/pluginmanager (127.0.0.1) 19.19ms
```
Which seems like a BC break in some python library, which should be easy to fix if I knew python. 

#### Did the same happen when running OctoPrint in safe mode?

No, safe mode is unable to do this by its nature AFAIK.

#### Version of OctoPrint

1.3.8 (from git)

#### Operating System running OctoPrint

As said, Debian 9.4 

#### Printer model & used firmware incl. version
Nothing to do with, printer unplugged. 

#### Browser and version of browser, operating system running browser
Mozilla Firefox 60.2 on Arch Linux. 

#### Link to octoprint.log
[https://gist.github.com/Valicek1/1b6945c72f596f4bd09abaadbea2296f](https://gist.github.com/Valicek1/1b6945c72f596f4bd09abaadbea2296f)
There is also attached output of `pip list` 

#### Link to contents of terminal tab or serial.log
Nothing to do with.

#### Link to contents of Javascript console in the browser
[https://gist.github.com/Valicek1/2137697f6596e9a4dd89c12d8f3b1f42](https://gist.github.com/Valicek1/2137697f6596e9a4dd89c12d8f3b1f42)

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ and also done a lot of googling.

foosel, 2018-06-20:
Looks like this is caused by a [backwards incompatible change introduced in sarge 0.1.5](http://sarge.readthedocs.io/en/latest/overview.html#id1) (compared to 0.1.4 which was current until yesterday). So much for semantic versioning ;) 

Thanks for reporting this, I'll see I can fix this still in time for 1.3.9rc1 which I actually was just in the process of testing.

foosel, 2018-06-20:
Should be solved by d3871e34873aa55476a8173859239a1a490fa279 which I've even pushed to `master` (so if you install from git, that should work out for you now). Will also be fixed in 1.3.9.

ClusterM, 2018-06-22:
Is it the same issue?
```
2018-06-23 02:15:23,098 - octoprint.plugins.pluginmanager - ERROR - Could not install plugin from https://github.com/fabianonline/OctoPrint-Telegram/archive/stable.zip
Traceback (most recent call last):
  File "/usr/share/octoprint/venv/lib/python2.7/site-packages/OctoPrint-1.3.8-py2.7.egg/octoprint/plugins/pluginmanager/__init__.py", line 348, in command_install
    returncode, stdout, stderr = self._call_pip(pip_args)
  File "/usr/share/octoprint/venv/lib/python2.7/site-packages/OctoPrint-1.3.8-py2.7.egg/octoprint/plugins/pluginmanager/__init__.py", line 652, in _call_pip
    return self._pip_caller.execute(*args)
  File "/usr/share/octoprint/venv/lib/python2.7/site-packages/OctoPrint-1.3.8-py2.7.egg/octoprint/util/pip.py", line 172, in execute
    return self.call(command, **kwargs)
  File "/usr/share/octoprint/venv/lib/python2.7/site-packages/OctoPrint-1.3.8-py2.7.egg/octoprint/util/commandline.py", line 83, in call
    p = sarge.run(command, **kwargs)
  File "/usr/share/octoprint/venv/lib/python2.7/site-packages/sarge-0.1.5.post0-py2.7.egg/sarge/__init__.py", line 1462, in run
    p.run(input=input, async_=async_)
  File "/usr/share/octoprint/venv/lib/python2.7/site-packages/sarge-0.1.5.post0-py2.7.egg/sarge/__init__.py", line 1071, in run
    self.run_node(node, input=input, async_=False)
  File "/usr/share/octoprint/venv/lib/python2.7/site-packages/sarge-0.1.5.post0-py2.7.egg/sarge/__init__.py", line 1187, in run_node
    result = getattr(self, method)(node, input, async_)
  File "/usr/share/octoprint/venv/lib/python2.7/site-packages/sarge-0.1.5.post0-py2.7.egg/sarge/__init__.py", line 1333, in run_command_node
    node.cmd.run(input=input, async_=async_)
  File "/usr/share/octoprint/venv/lib/python2.7/site-packages/sarge-0.1.5.post0-py2.7.egg/sarge/__init__.py", line 655, in run
    self.process = p = Popen(self.args, **self.kwargs)
TypeError: __init__() got an unexpected keyword argument 'async'
```

I still can install plugins using pip from command line. Version 1.3.8,

ClusterM, 2018-06-24:
@foosel Plugin installation fixed in 1.3.9rc1 but I still can't enable/disable plugins:
```
2018-06-25 02:27:42,723 - octoprint - ERROR - Exception on /api/plugin/pluginmanager [POST]
Traceback (most recent call last):
  File "/usr/share/octoprint/venv/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/share/octoprint/venv/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/share/octoprint/venv/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/share/octoprint/venv/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/share/octoprint/venv/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/usr/share/octoprint/venv/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1140, in decorated_view
    return flask_login.login_required(func)(*args, **kwargs)
  File "/usr/share/octoprint/venv/lib/python2.7/site-packages/Flask_Login-0.2.11-py2.7.egg/flask_login.py", line 758, in decorated_view
    return func(*args, **kwargs)
  File "/usr/share/octoprint/venv/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 99, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/usr/share/octoprint/venv/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 309, in on_api_command
    return self.command_toggle(plugin, command)
  File "/usr/share/octoprint/venv/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 577, in command_toggle
    self._mark_plugin_disabled(plugin, needs_restart=needs_restart)
  File "/usr/share/octoprint/venv/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 690, in _mark_plugin_disabled
    disabled_list = list(self._settings.global_get(["plugins", "_disabled"]))
TypeError: 'NoneType' object is not iterable
```

foosel, 2018-06-25:
This looks like you have something that doesn't belong in your disabled plugin list in `config.yaml`.

Please

  1. shutdown OctoPrint
  2. open `config.yaml` in a text editor
  3. delete the `plugins._disabled` line that is probably in there that looks like this:
      ``` yaml
      plugins:
        _disabled: null
      ```

If you *don't* have this in `config.yaml`, we'll have to dig deeper in a new ticket.

mexusbg, 2018-07-07:
I can confirm downgrading sarge 0.1.5 to 0.1.4 fix the issue for me! Thanks @foosel

gdachs, 2018-07-15:
at least for me it was not obvious how to downgrade my installation that I had made according to the documentation here: https://discourse.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian/2337. I am not a python programmer and don't know pip very well.
At least I could solve it by using this commands:
`cd ~/OctoPrint
source venv/bin/activate
pip install sarge==0.1.4
`
Maybe it can help someone

foosel, 2018-12-10:
Forgot to close this on release of 1.3.9

closed 2018-12-10 by foosel