OctoPrint/OctoPrint #2689: Drag and drop isn't working with current devel branch

koenkooi, 2018-06-21:
#### What were you doing?

Trying to drag & drop a gcode file onto octoprint. It wasn't working with 7b773f6fd21d407b325ce5f507bec24b945b9e21,  updating to e58387b4f2df0c02776a8d2f08265fa9ab84ba1f makes no difference.

#### What did you expect to happen?

Octoprint to show d&d overlay.

#### What happened instead?

No overlay.

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Version of OctoPrint

Commit e58387b from devel branch,  ```OctoPrint 1.4.0.dev1313+ge58387b```

#### Operating System running OctoPrint

Linux 4.14.7 on a beaglebone black
Linux 4.14.22 on a raspberrypi2

#### Printer model & used firmware incl. version

Prusa i3 mk3 running 3.2.1
Fabrikator mini running marlin bugfix-2.0

#### Browser and version of browser, operating system running browser

Chrome 66.0.3359.181 on OSX
Safari 11.1.1 (13605.2.6) on OSX

#### Link to octoprint.log

https://gist.github.com/koenkooi/f1762d4544c5f6585b9d862360a9bac0

#### Link to contents of Javascript console in the browser

[image]

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

gege2b, 2018-06-21:
Actually it doesn't work since at least may 14th because I spotted this bug when I opened my issue about the update button not worling

I just forgot to post this issue since then

Same issue with firefox 60.0.2 64bit on win10

foosel, 2018-06-21:
Sounds like a merge went wrong somewhere. I'll take a look

foosel, 2018-06-22:
Actually, that was a refactoring that went wrong, and even before the permission branch was merged. Thanks for reporting this, should be fixed by the above commit!

closed 2018-06-22 by foosel

koenkooi, 2018-06-22:
Fix confirmed!

commit 38ddd310 (2018-06-22): Fix another broken permission

Closes #2689