OctoPrint/OctoPrint #2690: [Request]  Please add a way to upload from OctoPrint to SD.

AndKe, 2018-06-21:
Today, I can send a job from Slic3r to OctoPrint,  but if I want it on printer's SD card, I have to download it, then click "upload to SD" 
Please allow uploading directly from OctoPrint's filelist.

tedder, 2018-06-24:
It's possible both manually (eg drag it onto the right half of the octoprint screen) and via the API ([by setting the location to sdcard`](http://docs.octoprint.org/en/master/api/files.html#upload-file-or-create-folder)). It just needs to be supported in slic3r.

tedder, 2018-06-24:
In fact, looks like you might be able to monkeypatch slic3r pretty easily [here](https://github.com/slic3r/Slic3r/blob/c058ae91d245a3de5e3ff9388439d98a0da69434/lib/Slic3r/GUI/Plater.pm#L2273) and [here](https://github.com/slic3r/Slic3r/blob/c058ae91d245a3de5e3ff9388439d98a0da69434/lib/Slic3r/GUI/Plater.pm#L2319) as a workaround.

Looks like [a patch to enable this was proposed](https://github.com/slic3r/Slic3r/pull/2623) a few years ago.