OctoPrint/OctoPrint #2691: M42 S0-S255 Slider Plugin

Stevetecx, 2018-06-22:
Is it possible that someone generates a plugin that sends an M42 command from S0 to S255 via slider.
Or by keys 0% - 25% - 50% - 75% - 100%

I use the M42 for light control on my Ultimaker 2+. Good thing but cumbersome to enter a command each time.

Thanks

Stevetecx, 2018-06-22:
request a feature

ntoff, 2018-06-22:
For reference, you need the title to start with [request] to keep the bot happy

and your request is super easy to do with custom controls https://github.com/foosel/OctoPrint/wiki/Cookbook:-Custom-Controls so no need for a plugin.

You could modify the fan speed custom control to send out your M42 command.

foosel, 2018-06-22:
As @ntoff mentioned, no need for a plugin here, some custom controls are enough :)

closed 2018-06-22 by foosel