OctoPrint/OctoPrint #2693: Temperature randomly set to 360 degrees

ClusterM, 2018-06-23:
#### What were you doing?

Trying to print, my G-code:
```
; generated by Slic3r 1.3.1-dev on 2018-06-23 at 02:01:18

; external perimeters extrusion width = 0.33mm (2.01mm^3/s)
; perimeters extrusion width = 0.35mm (4.24mm^3/s)
; infill extrusion width = 0.35mm (5.65mm^3/s)
; solid infill extrusion width = 0.35mm (1.41mm^3/s)
; top infill extrusion width = 0.35mm (1.06mm^3/s)

M107
M190 S90 ; set bed temperature and wait for it to be reached
M104 S230 ; set temperature
M107 ; fan off
M42 P30 S255 ; heatbreaker fan on
G28 ; home all axes
G1 Z5 F5000 ; lift nozzle

M109 S230 ; set temperature and wait for it to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
---and so on---
```

I uploaded file using Slic3r and started printing using "Print" button.

#### What did you expect to happen?

Target temperature of hotend should be 230 degrees.

#### What happened instead?

For some reason "M104 S360.000000" is sent instead of "M104 S230".

serial.log:
```
2018-06-23 09:25:36,612 - Changing monitoring state from "Operational" to "Printing"
2018-06-23 09:25:36,680 - Send: N0 M110 N0*125
2018-06-23 09:25:36,695 - Recv: ok
2018-06-23 09:25:36,712 - Send: N1 M107*36
2018-06-23 09:25:36,728 - Recv: ok
2018-06-23 09:25:36,743 - Send: N2 M190 S90*83
2018-06-23 09:25:36,759 - Recv:  T:27.40 /0.00 B:28.00 /90.00 @:0 B@:0 W:?
---SKIPPED---
2018-06-23 09:30:29,788 - Recv:  T:28.60 /0.00 B:89.62 /90.00 @:0 B@:127 W:0
2018-06-23 09:30:30,604 - Recv: ok
2018-06-23 09:30:30,626 - Send: N3 M104 S360.000000*77
2018-06-23 09:30:30,637 - Recv: ok
2018-06-23 09:30:30,676 - Send: N4 M107*33
2018-06-23 09:30:30,684 - Recv: ok
2018-06-23 09:30:30,694 - Send: N5 M42 P30 S255*34
2018-06-23 09:30:30,716 - Recv: ok
2018-06-23 09:30:30,734 - Send: N6 G28*21
2018-06-23 09:30:31,084 - Recv:  T:28.80 /360.00 B:89.77 /90.00 @:127 B@:127
---SKIPPED---
2018-06-23 09:30:49,095 - Recv:  T:64.64 /360.00 B:89.00 /90.00 @:127 B@:127
---SKIPPED---
2018-06-23 09:33:01,371 - Recv: echo:busy: processing
---SKIPPED---
2018-06-23 09:33:05,115 - Recv:  T:269.75 /360.00 B:90.49 /90.00 @:127 B@:0
2018-06-23 09:33:05,371 - Recv: echo:busy: processing
2018-06-23 09:33:05,387 - Recv:  T:269.50 /360.00 B:90.57 /90.00 @:127 B@:0 W:?
2018-06-23 09:33:05,958 - Recv: Error:MAXTEMP triggered, system stopped! Heater_ID: 0
2018-06-23 09:33:05,988 - Changing monitoring state from "Printing" to "Cancelling"
2018-06-23 09:33:06,023 - Recv: Error:Printer halted. kill() called!
```

#### Did the same happen when running OctoPrint in safe mode?

It happens randomly, so I don't know. It has not happened yet.

#### Version of OctoPrint

1.3.8

#### Operating System running OctoPrint

```
root@3Dprinter:~# uname -a
Linux 3Dprinter 4.4.74 #0 Tue Mar 27 17:21:32 2018 mips GNU/Linux
```

#### Printer model & used firmware incl. version

RepRap Prusa Mendel clone
Marlin 1.1.8

#### Browser and version of browser, operating system running browser

Google Chrome | 67.0.3396.87 (x64) (cohort: Stable)
Version | 878cd31214ac27a3996927cd5c9c138b10c9fc8d-refs/branch-heads/3396@{#771}
OS | Windows
JavaScript | V8 6.7.288.46

Also happened with Google Chrome for Android.

#### Link to octoprint.log
octoprint.log: https://pastebin.com/XxML4073

#### Link to contents of terminal tab or serial.log

serial.log: https://pastebin.com/FFuM9U9n

#### Link to contents of Javascript console in the browser

Last time problem appeared using mobile browser, there is no Javascript console, sorry.

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

ClusterM, 2018-06-23:
@ntoff You can check my g-code: http://clusterrr.com/temp/camera_mount.gcode
There is no lines about 360 degrees. It prints normally without Octoprint. Also it prints normally if I just restart printing without any changes of g-code.

ntoff, 2018-06-23:
You have a TONNE of plugin errors, and pip 10 which is known to be broken. https://discourse.octoprint.org/t/installing-a-plugin-shows-me-unknown-in-the-popup-i-cant-uninstall-plugins-through-the-plugin-manager-octoprint-up-to-1-3-8-octopi-0-15-0-or-pip-10-x/1746

I would try downgrading pip and work on fixing those plugin errors. You say it hasn't happened in safe mode yet so it's possible one of those broken plugins is causing issues.

ClusterM, 2018-06-23:
I tried to install plugin today, it produced those errors,i just wrote about it there: https://github.com/foosel/OctoPrint/issues/2687#issuecomment-399609137
So maybe causes problem.
I'll downgrade pip.

ntoff, 2018-06-24:
Any luck fixing the plugin errors? If so, did it also fix your random temperature spikes?

Also one other thing to check, make sure you haven't accidentally set any sort of temperature offset  in octoprint.

ClusterM, 2018-06-24:
I have done many printing in safe mode today without any problems. So seems like it's really plugin problem. I'll test 1.3.9rc1 with plugins. Will close issue for now.

closed 2018-06-24 by ClusterM

ntoff, 2018-06-25:
I would test it with whatever you currently use, disable all plugins and enable them one by one. Don't change too many variables all at once. At least you  know your current version of octoprint itself is stable, then you  can work through the plugins to find the culprit (or maybe it was just the sarge / pip 10 issue causing it all along).

foosel, 2018-06-25:
Doubled temperature sounds like [this](https://github.com/foosel/OctoPrint/issues/2550). Might be the Filament Manager Plugin again.