OctoPrint/OctoPrint #2694: [Feature Request] Rename files in OctoPrint files list.

MichMich, 2018-06-23:
#### What were you doing?

1. Imported a model into Cura.
2. Connect Cura to OctoPrint Using API Key
3. Press "Print With Octoprint" button.

#### What did you expect to happen?

The file was send to Octoprint using a cryptic name `LM_254ffe25-7813-4345-96fa-d8b5c16a2144.gcode` because I forgot to rename it in Cura. I hoped I could rename this file in OctoPrint after I finished my print to keep my print archive ordered.

#### What happened instead?

OctoPrint did not offer an option to rename the files using the UI.

#### Did the same happen when running OctoPrint in safe mode?

Yes, it's a missing feature.

#### Version of OctoPrint

OctoPrint 1.3.8 running on OctoPi 0.14.0

#### Operating System running OctoPrint

OctoPi 0.14.0

#### Printer model & used firmware incl. version

NA.

#### Browser and version of browser, operating system running browser

NA.

#### Link to octoprint.log

NA.

#### Link to contents of terminal tab or serial.log

NA.

#### Link to contents of Javascript console in the browser

NA.

#### Screenshot(s)/video(s) showing the problem:

NA

I have read the FAQ.

ntoff, 2018-06-23:
https://plugins.octoprint.org/plugins/filemanager/ I'm not 100% sure but I think you can rename them with the filemanager plugin. Give it a go and see.

MichMich, 2018-06-24:
@ntoff You're correct. This plugin allows me to rename. Thanks for the suggestion!

I leave this issue open for @foosel to decide if she wants to incorporate this feature in the OctoPrint core. @foosel: feel free to close this issue without further message if you think the current solution (using a plugin) is sufficient.

foosel, 2018-06-25:
For now I'd say, use the plugin. I've been thinking about maybe bundling it in the future but so far haven't had a chance to take a closer look at that.

closed 2018-06-25 by foosel