OctoPrint/OctoPrint #2699: Octoprint detect PRINT DONE before the actual print end

Gi0cas, 2018-06-24:
#### What were you doing?

I was printing. I use Klipper firmware with a Orange Pi Zero.

I experienced this issue : 
Octoprint tells me that the print is finished but really isn't and in the Terminal i can see the flow of the remaining gcode (usually the last 2-3 layer of the print)...
In the last two prints, Octoprint told me that the print "ended" about 5 minutes before the actual end.

I read that Octoprint use "PrintDone" hook by Marlin to know when the print is finished, but i use Klipper and probably there is no PrintDone status that Octoprint can read...how can i set Octoprint to wait till the end of the gcode?

I need this, because i installed a system that cut the power after the print, and in this case Octoprint cutted the last top layers...

#### Did the same happen when running OctoPrint in safe mode?
i think it's not a problem solvable with safe mode

#### Version of OctoPrint
1.3.8

#### Operating System running OctoPrint
Orange pi zero Armbian

#### Printer model & used firmware incl. version

Custom printer, Kipper software

I have read the FAQ.

Gi0cas, 2018-06-24:
I closed the issue because probably the "printdone" hook excludes the code relative to end print. I can try to inject this end gcode in the gcode file to verify that octoprint can complete the potential totally

foosel, 2018-06-25:
> I read that Octoprint use "PrintDone" hook by Marlin to know when the print is finished

I don't know where you read this, but this isn't the case (at least not unless you are printing from SD).

OctoPrint until 1.3.8 will fire the `PrintDone` event and also switch away from "Printing" state when it has read the last line from the file it's currently sending to the printer. 

Starting with 1.3.9rc1 (just released last week) OctoPrint will instead send an `M400` at the end of the print to wait for all movements to finish, switch to a new "Finishing" state, then fire the `PrintDone` event, then run the "afterPrintDone" GCODE script, and only once THAT has been sent completely it will switch from "Finishing" to "Operational" again.

I still would only use this event or this state change to start a timer though, to make sure stuff is cooled down enough that no filament blockage can be caused by shutting off the fans and such. I think the [PSU Control Plugin](https://plugins.octoprint.org/plugins/psucontrol/) even does this already for you so you might want to take a look at that.

Gi0cas, 2018-06-25:
Ok Gina, thank you. I found so where is the problem...i'm using octoprint 1.3.8 and when Octoprint read the last line of the gcode, immediately fire the PrintDone, that immediately cut the power and the printer hasn't the time to complete this gcode just readed.

I take a look at this plugin, thank you again !