OctoPrint/OctoPrint #2700: Offline (Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:2417)

Electroprint, 2018-06-24:
-->

#### What were you doing?

The Serial Connection randomly disconnects during the print, i have tried the same print several times and each time it disconnects at a different point in the print (between 5 minutes and 1.5 hours). When using repetier host it has never disconnected. 

#### What did you expect to happen?

The Serial Connection to stay connected during the entire print

#### What happened instead?

Serial Connection disconnected and print stopped/failed

#### Did the same happen when running OctoPrint in safe mode?

Same issue in safe mode

#### Version of OctoPrint

OctoPrint 1.3.8

#### Operating System running OctoPrint

OctoPi 0.15.1

#### Printer model & used firmware incl. version

Unknown version of Marlin (but no issues with repetier host ever)

#### Browser and version of browser, operating system running browser

N/A

#### Link to octoprint.log

https://pastebin.com/ecuhVdKx

#### Link to contents of terminal tab or serial.log

https://pastebin.com/MCRc3hb3

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s)/video(s) showing the problem:

N/A

I have read the FAQ.

foosel, 2018-06-25:
Please see [this FAQ entry](https://discourse.octoprint.org/t/octoprint-randomly-loses-connection-to-the-printer-with-a-serialexception/228). In short: It's something beneath OctoPrint breaking away and there's nothing that OctoPrint can do about it. A number of common reasons for this are listed in the FAQ entry.

closed 2018-06-25 by foosel

mattfaraday, 2020-05-28:
??? how can you determine that it's not octoprint. 

It worked before upgrade, it failed afterwards. The cable was replaced too.