OctoPrint/OctoPrint #2703: Bed and hot end not being told to heat up

CarlRowbs, 2018-06-25:
My problem is when setting print it does not set bed temp or hot end just starts trying to print and on top of not heating it's not starting the motor to put the filament out 

Changing monitoring state from "Operational" to "Printing"
Send: N16 M107*18
Recv: ok
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115 U3.1.1-RC5*104
Recv: ok
Send: N2 M83*26
Recv: ok
Send: N3 G28 W*103
Recv: tmc2130_home_enter(axes_mask=0x01)
Recv:   0 step=61 mscnt= 990
Recv: tmc2130_goto_step 0 61 2 1000
Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x01
Recv: tmc2130_home_enter(axes_mask=0x02)
Recv: echo:busy: processing
Recv:   0 step=49 mscnt= 798
Recv: tmc2130_goto_step 1 49 2 1000
Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x02
Recv: ok
Send: N4 M400*35
Recv: ok
Send: N5 M114*34
Recv: X:12.00 Y:6.00 Z:0.15 E:0.00 Count X: 11.47 Y:6.70 Z:0.15 E:0.00
Recv: ok
Send: N6 G80*23
Recv: echo:busy: processing
Recv: echo:busy: processing
Changing monitoring state from "Printing" to "Cancelling"
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing

ntoff, 2018-06-26:
Your gcode appears to have no heatup command, please use the forum for help and support https://discourse.octoprint.org/ on troubleshooting your gcode commands. 

This issue tracker is reserved for bug reports, if you believe there to be an actual bug, please fill out the ticket in its entirety following the template https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report and make sure you include the serial log (serial.log is required to troubleshoot issues between printer and host).

foosel, 2018-06-26:
As already mentioned, for questions or other kind of help or support, please get in touch on the [community forum](https://discourse.octoprint.org). The bug tracker is reserved for bugs, feature requests and brainstorming tickets. See also the [Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md). Thank you! :)

closed 2018-06-26 by foosel