OctoPrint/OctoPrint #2704: Timelapse URI test not working on 'devel' branch

koenkooi, 2018-06-26:
#### What were you doing?
I was trying to change the timelapse URI to point to the proxy instead of mjpg-streamer to fix the conflict between the timelapse and pushover trying to access it at the same time at the end of a build.

#### What did you expect to happen?

The test picture to pop up.

#### What happened instead?

```
Jun 25 14:33:01 blackbox octoprint[28754]: [2018-06-25 14:33:01,700] ERROR in app: Exception on /api/util/test [POST]
Jun 25 14:33:01 blackbox octoprint[28754]: Traceback (most recent call last):
Jun 25 14:33:01 blackbox octoprint[28754]:   File "/usr/lib/python2.7/site-packages/flask/app.py", line 1982, in wsgi_app
Jun 25 14:33:01 blackbox octoprint[28754]:     response = self.full_dispatch_request()
Jun 25 14:33:01 blackbox octoprint[28754]:   File "/usr/lib/python2.7/site-packages/flask/app.py", line 1614, in full_dispatch_request
Jun 25 14:33:01 blackbox octoprint[28754]:     rv = self.handle_user_exception(e)
Jun 25 14:33:01 blackbox octoprint[28754]:   File "/usr/lib/python2.7/site-packages/flask/app.py", line 1517, in handle_user_exception
Jun 25 14:33:01 blackbox octoprint[28754]:     reraise(exc_type, exc_value, tb)
Jun 25 14:33:01 blackbox octoprint[28754]:   File "/usr/lib/python2.7/site-packages/flask/app.py", line 1612, in full_dispatch_request
Jun 25 14:33:01 blackbox octoprint[28754]:     rv = self.dispatch_request()
Jun 25 14:33:01 blackbox octoprint[28754]:   File "/usr/lib/python2.7/site-packages/flask/app.py", line 1598, in dispatch_request
Jun 25 14:33:01 blackbox octoprint[28754]:     return self.view_functions[rule.endpoint](**req.view_args)
Jun 25 14:33:01 blackbox octoprint[28754]:   File "/usr/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1158, in decorated_view
Jun 25 14:33:01 blackbox octoprint[28754]:     return func(*args, **kwargs)
Jun 25 14:33:01 blackbox octoprint[28754]:   File "/usr/lib/python2.7/site-packages/flask_principal.py", line 199, in _decorated
Jun 25 14:33:01 blackbox octoprint[28754]:     rv = f(*args, **kw)
Jun 25 14:33:01 blackbox octoprint[28754]:   File "/usr/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 264, in utilTest
Jun 25 14:33:01 blackbox octoprint[28754]:     return _test_url(data)
Jun 25 14:33:01 blackbox octoprint[28754]:   File "/usr/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 420, in _test_url
Jun 25 14:33:01 blackbox octoprint[28754]:     content_type_whitelist = map(util.parse_mime_type, data["content_type_whitelist"])
Jun 25 14:33:01 blackbox octoprint[28754]: NameError: global name 'util' is not defined
Jun 25 14:33:01 blackbox octoprint[28754]: 2018-06-25 14:33:01,735 - tornado.access - ERROR - 500 POST /api/util/test (fe80::4bc:80e3:29bd:2adb) 142.06ms
Jun 25 18:33:53 blackbox octoprint[28754]: 2018-06-25 18:33:53,839 - tornado.access - WARNING - 404 GET /api/printerprofiles/ (172.20.0.251) 32.74ms
```

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

OctoPrint 1.4.0.dev1315+g38ddd31
OctoPrint 1.4.0.dev1317+g033e567

#### Operating System running OctoPrint

Linux kernel 4.14.7 on a beaglebone black
Linux kernel 4.17.3 on a beaglebone black

#### Printer model & used firmware incl. version

Fabrikator mini, marlin bugfix-2.0

#### Browser and version of browser, operating system running browser

Safari 11.1.1 (13605.2.6) on OSX 10.13.5 bèta (17F59b)

#### Link to octoprint.log

https://gist.github.com/koenkooi/dae3629d3f434bf216b170cc7061ee79

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

koenkooi, 2018-07-04:
Fixed by 88a962b3ab1e0b162dade8239ef51f1e19ca9229

koenkooi, 2018-07-04:
[image]

closed 2018-07-04 by koenkooi

foosel, 2018-07-04:
Hah, thanks for that! ❤ I knew I'd seen a ticket about this somewhere but hadn't yet gotten a chance to check.