OctoPrint/OctoPrint #2709: Question/Suggestion: Printer variants

Thisismydigitalself, 2018-06-30:
Hi.
I will start by saying that only recently i moved my Start/end codes from my Slicer to Octoprint and I love this as it makes much more sense. Now to my question:  I need to print on a given printer but sometimes with different start/end codes. is it possible to configure aliases for that? I guess i can create different printers for each start/end codes, but it will be easier to link a given printer to a different set of commands followed by OctoPrint issuing a new tag/flag suffix, thus preserving its API key. something like:

PrinterA.SetupX
PrinterA.SetupY
PrinterB.SetupZ

And so on.
My slicer (Cura) should then present these options to the user to chose from as these were different printers.

Just thinking out loud.
Thank you. 

OctoPrint - I'm lovin' it!

foosel, 2018-07-06:
Hm... this would increase complexity significantly. It might already be possible through a plugin which extends the existing GCODE scripts on a per print basis, but I'm not going to build something like this into the core. 

> only recently i moved my Start/end codes from my Slicer to Octoprint and I love this as it makes much more sense.

To be honest, the Start/End GCODE scripts in OctoPrint are not meant as a *replacement* of the ones in your slicer but rather as *additional* stuff to do before/after prints. If it works for you though, great.

Thisismydigitalself, 2018-07-20:
Thank you.
While creating new printer for me to set different Start/End codes for each I realized that the codes are global for all printers. Am i missing something?

This is wrong.

foosel, 2018-07-20:
As I said, the Start/End GCODE scripts in OctoPrint are not meant as a replacement of the ones in your slicer but rather as additional stuff to do before/after prints, so *OctoPrint specific code*, not printer specific code.

If someone wants to add printer profile specific Start/End GCODE scripts, I'm happy to accept PRs.