OctoPrint/OctoPrint #2711: Octoprint Scaling Up Objects After Loading

Quintin9801, 2018-07-02:
#### What were you doing? Loading in a model of a 20x20x20mm calibration cube, estimation in cura is 18min and that's normal for the amount of infill I had set, I've printed this model multiple times for new nozzles BUT when I save it to my OctoPrint Watched folder and load it up it told me the print would take 2 days!

1. Load up model in cura and save
2. Load up in Octoprint Interface
3. BAM freak accident that has unreal estimates and when you try to print it, it just goes off the print bed.

#### What did you expect to happen? 
Normal loading and printing for calibration
#### What happened instead?
Insane up-scaling of model
#### Did the same happen when running OctoPrint in safe mode?
Yes, as shown in the screenshot, the model is still huge in safe mode and Ive tried changing my cura settings as well

#### Version of OctoPrint
OctoPrint 1.3.8

#### Operating System running OctoPrint
 running on OctoPi 0.15.1
Cura version 3.4.0

#### Printer model & used firmware incl. version
Anet A8 

#### Browser and version of browser, operating system running browser
Google Crome Version 67.0.3396.99 (Official Build) (64-bit)

#### Link to octoprint.log
[octoprint.log](https://pastebin.com/ii2aj7NQ)

#### Link to contents of terminal tab or serial.log
Not Enabled, sorry
[serial.log](https://pastebin.com/JVWDbKqk)

#### Link to contents of Javascript console in the browser
[Browser JavaScript](https://pastebin.com/iHdUveUu)

#### Screenshot(s)/video(s) showing the problem:
[image]
[image]
[image]

I have read the FAQ.

ntoff, 2018-07-03:
So you're slicing externally using Cura, not the built in cura slicer ability in octoprint? (just making sure)

If you could please upload your gcode for that cube, that would probably assist in any attempts to replicate /test this.

foosel, 2018-07-03:
OctoPrint can't "scale up" already sliced models. For that it would need to have access to the source STL and slice again. Please share an affected GCODE file as @ntoff already requested, I'm fairly sure that this will show that the generated GCODE from your slicer is in fact the issue here, either by actually being this big, or by confusing the GCODE analysis into *believing* it is this big.

Quintin9801, 2018-07-03:
@ntoff correct I am using cura externally to octoprint and this link has the .gcode from the Octoprint interface, as well as the .gcode from Cura and the original .stl

https://drive.google.com/open?id=1GJ639U9dhRfA6v2-94t4l-cN8BZpy3ea

foosel, 2018-07-03:
So this is a problem with the GCODE parser.

Your start code sets relative X/Y/Z positioning via `G91` but then never switches back to absolute positioning via `G90`:

```
;FLAVOR:Marlin
;TIME:1123
;Filament used: 1.37114m
;Layer height: 0.2
;Generated with Cura_SteamEngine 3.4.0
M190 S40
M104 S195
M109 S195
M82 ;absolute extrusion mode
;Sliced at: Tue 03-07-2018 09:04:40
G21        ;metric values
G90        ;absolute positioning
M82        ;set extruder to absolute mode
M107       ;start with the fan off
G28 X0 Y0  ;move X/Y to min endstops
G91        ;Relative positioning
G1 Z5 F200           ;Raise Z axis 5mm
G0 X110 Y110 F7500   ;Center build plate
G28 Z0  F200               ;Home Z axis
G29        ;auto home
G1 Z15.0 F200 ;move the platform down 15mm
G92 E0                  ;zero the extruded length
G1 F200 E15              ;extrude 15mm of feed stock
G92 E0                  ;zero the extruded length again
G4 S5
G1 F{travel_speed}
M117 Printing... ;Put printing message on LCD screen
G92 E0
G1 F1500 E-6.5
```

All of the moves look like they are sliced with absolute coordinates though. My guess is your firmware interprets any of the following commands a reset to absolute coordinates. There's nothing in there though that does this explicitly (btw, there's also a left over placeholder here that your slicer didn't properly swap out, you want to look into this: `G1 F{travel_speed}`), which makes OctoPrint think all of the moves are relative, which leads to those ridiculous numbers.

My guess is, if you add a `G90` to the end of your start GCODE, OctoPrint's GCODE viewer and analysis will also behave.

What I'm unclear of right now is which of the commands in there also contains an implicit switch to absolute coordinates - once I know that I can see if I can adjust the parser routines in OctoPrint to accomodate files like that.

In any case, it's only the analysis and reporting that's hiccuping here - if you were to print this (and your firmware *actually* switches back to absolute positioning!) it should work.

Quintin9801, 2018-07-03:
Thank you so much @foosel i went through many troubleshooting steps and it makes sense that I overthought such a simple change that I made, I appreciate your understanding and for everything you’ve done for octoprint I use it every day and makes printing so much more automated and easy!

closed 2018-07-03 by Quintin9801

foosel, 2018-07-03:
Just for the record: Simplify3D's GCODE viewer also renders this wrongly:

[image]

That extrusion line there running towards the horizon is part of the first layer ;)