OctoPrint/OctoPrint #2714: OctoPrint-Telegram 1.3.9rc1 cant install Telegram plugin

seraserboy, 2018-07-04:
#### What were you doing?

1. ...
2. ...
3. ...

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-07-04:
```
[...]
2018-07-05 03:01:43,026 - octoprint.plugin.core - ERROR - Error loading plugin telegram
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 847, in _import_plugin
    instance = imp.load_module(key, f, filename, description)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_telegram/__init__.py", line 2, in <module>
    from PIL import Image
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/PIL/Image.py", line 64, in <module>
    from . import _imaging as core
ImportError: /usr/lib/arm-linux-gnueabihf/libc.so: invalid ELF header
[...]
```

That looks like an issue outside of the control of OctoPrint. Either it's your environment that's the problem here, or the plugin trying to pull in a wrong library version. I'd suggest to report this with the plugin author, just to be sure, but I fear it's more a case of your environment - for some reason the PIL dependency is trying to load a binary with a header that doesn't fit your operating system/processor.

closed 2018-07-04 by foosel

seraserboy, 2018-07-04:
Ok, but you can Install this plugin un 1.3.9rc1 fine?

foosel, 2018-07-05:
Yep!

```
Installing plugin "OctoPrint-Telegram" from https://github.com/fabianonline/OctoPrint-Telegram/archive/stable.zip...
/home/pi/oprint/bin/python2 -m pip install https://github.com/fabianonline/OctoPrint-Telegram/archive/stable.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/fabianonline/OctoPrint-Telegram/archive/stable.zip
Downloading https://github.com/fabianonline/OctoPrint-Telegram/archive/stable.zip (98kB)
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-Telegram==1.4.2) (1.3.9rc1)
Collecting pillow (from OctoPrint-Telegram==1.4.2)
Downloading https://files.pythonhosted.org/packages/d3/c4/b45b9c0d549f482dd072055e2d3ced88f3b977f7b87c7a990228b20e7da1/Pillow-5.2.0.tar.gz (14.5MB)
Requirement already satisfied: watchdog<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (0.8.3)
Requirement already satisfied: python-dateutil<2.7,>=2.6.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (2.6.1)
Requirement already satisfied: feedparser<5.3,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (5.2.1)
Requirement already satisfied: requests<3,>=2.18.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (2.18.4)
Requirement already satisfied: flask<0.11,>=0.10.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (0.10.1)
Requirement already satisfied: scandir<1.4,>=1.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (1.3)
Requirement already satisfied: monotonic<1.4,>=1.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (1.3)
Requirement already satisfied: frozendict<1.3,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (1.2)
Requirement already satisfied: Jinja2<2.9,>=2.8.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (2.8.1)
Requirement already satisfied: emoji<0.5,>=0.4.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (0.4.5)
Requirement already satisfied: netifaces<0.11,>=0.10.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (0.10.6)
Requirement already satisfied: Flask-Babel<0.10,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (0.9)
Requirement already satisfied: pkginfo<1.5,>=1.4.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (1.4.2)
Requirement already satisfied: markdown<2.7,>=2.6.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (2.6.11)
Requirement already satisfied: pyserial<3.5,>=3.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (3.4)
Requirement already satisfied: semantic-version<2.5,>=2.4.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (2.4.2)
Requirement already satisfied: werkzeug<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (0.8.3)
Requirement already satisfied: websocket-client<0.41,>=0.40.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (0.40.0)
Requirement already satisfied: psutil<5.5,>=5.4.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (5.4.6)
Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (1.6.5)
Requirement already satisfied: Flask-Login<0.3,>=0.2.11 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (0.2.11)
Requirement already satisfied: Flask-Principal<0.4,>=0.3.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (0.3.5)
Requirement already satisfied: rsa<3.3,>=3.2.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (3.2.3)
Requirement already satisfied: pylru<1.1,>=1.0.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (1.0.9)
Requirement already satisfied: Flask-Assets<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (0.10)
Requirement already satisfied: netaddr<0.8,>=0.7.19 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (0.7.19)
Requirement already satisfied: future<0.16,>=0.15.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (0.15.2)
Requirement already satisfied: Click<6.3,>=6.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (6.2)
Requirement already satisfied: sarge==0.1.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (0.1.4)
Requirement already satisfied: chainmap<1.1,>=1.0.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Telegram==1.4.2) (1.0.2)
[... cut ...]
2018-07-05 13:26:52,216 - octoprint.environment - INFO - Detected environment is Python 2.7.13 under Linux (linux2). Details:
|  hardware:
|    cores: 4
|    freq: 1400.0
|    ram: 918192128
|  os:
|    id: linux
|    platform: linux2
|  plugins:
|    octopi_support:
|      model: unknown
|      revision: a020d3
|      version: 0.15.0
|  python:
|    pip: 10.0.1
|    version: 2.7.13
|    virtualenv: /home/pi/oprint
```

I find it interesting though that it installed pillow for me, but apparently PIL for you.

seraserboy, 2018-07-05:
Thank you Gina.

Finally working after some hours of Google and test.

cd /lib/arm-linux-gnueabihf
sudo ln -s libc.so.6 libc.so

This is the solution.