OctoPrint/OctoPrint #2716: [Request] Default to autoscroll in terminal tab

Ellmen, 2018-07-05:
Currently you have to specifically click autoscroll to get a continuous feed from the terminal tab. It strikes me that this would be the expected default behaviour. Also, when the browser window loses focus (i.e. switching tabs), autoscroll gets disabled again which is kind of annoying.

I have read the FAQ.

ntoff, 2018-07-06:
Really? I'm pretty sure it's always auto scrolled  by  default for me, do you have the auto scroll plugin installed by any chance? The one that turns off auto scrolling if you manually scroll.

foosel, 2018-07-13:
It's auto scrolling just fine here in Chrome (but on Windows, not OSX).

Please test if the issue persists in safe mode. If not it's one of your installed third party plugins that's causing problems. If yes fill out [the ticket template](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report) and ideally include a video that shows the wrong behaviour.

foosel, 2018-07-25:
Closing due to lack of response. Feel free to request a reopen once the requested information is available.

closed 2018-07-25 by foosel