OctoPrint/OctoPrint #2719: [1.3.9rc2] Estimator can stay uninitialized - doesn't cause an error, but also prevents print time left calculation

foosel, 2018-07-06:
### Problem

As reported by @ctgreybeard in https://github.com/foosel/OctoPrint/issues/2718#issuecomment-403047866 the print time estimation can stay at "Still stabilizing"/`None` endlessly.

This is caused by a wrong logical order in `on_comm_file_selected`, causing the estimator to initialize before the current print job data is set (which is needed for this). Depending on locking it can work or fail, which is probably the reason it wasn't detected during initial testing.

### Solution

Fix initialization order.

OutsourcedGuru, 2018-07-16:
I don't know if it's related but I'm currently gathering info/evidence for either 1) an OctoPrint issue or one for 2) the PrintTimeGenius plugin itself. The print job that's running now should help figure out which is the cause. I'm now running without that plugin to make the determination.

What: v1.3.9rc2
Description: Finished job is in state "Finishing" forever

[image]

OutsourcedGuru, 2018-07-17:
Ignore me. I have a feeling that I was causing this myself.

My gcode scripts when the job finishes runs an OCTO806 which in terms is connected to `aplay filename`. This appears to be a blocking call and since the audio lasts perhaps five seconds, it seems to mess up the finishing routine.

I've adjust the command to be `aplay filename &` (putting the command in the background).

foosel, 2018-07-20:
Ok, ignoring ;)

Also just realized that I forgot to close this now that rc3 (and in a couple minutes even rc4) is out.

commit 386ae760 (2018-07-06): Fix initialization order of estimator

Closes #2719
closed 2018-07-20 by foosel