OctoPrint/OctoPrint #2720: mk3 bed leveling failed leads to disconnect

tedder, 2018-07-06:
Dumping the terminal output here. Maybe this is correct (that it should disconnect), but from a UX perspective it seems like presenting some sort of error to the user would be better.

```
Recv: T:215.5 E:0 W:0
Recv: ok
Send: N15 G28 W*80
Recv: tmc2130_home_enter(axes_mask=0x01)
Recv: echo:busy: processing
Printer seems to support the busy protocol, adjusting timeouts and setting busy interval accordingly
Recv:   0 step=29 mscnt= 479
Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x01
Recv: tmc2130_home_enter(axes_mask=0x02)
Recv: echo:busy: processing
Recv:   0 step=18 mscnt= 290
Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x02
Recv: echo:busy: processing
Recv: ok
Send: N16 M113 S2*86
Recv: ok
Send: N17 G80*39
Recv: echo:busy: processing
Recv: KILL: 0
Recv: Error:Printer halted. kill() called!
Changing monitoring state from "Printing" to "Error: Printer halted. kill() called!"
Changing monitoring state from "Error: Printer halted. kill() called!" to "Offline (Error: Printer halted. kill() called!)"
Connection closed, closing down monitor
```

chatrat12, 2018-07-07:
@tedder I opened a similar ticket. I just updated my OctoPrint the latest release candidate (1.3.9rc2) and I'm not having issues with it anymore. Can you try the same and see if it fixes your issue?

chatrat12, 2018-07-07:
I just got this error on my printer cause didn't have the build plate on correctly. Got the same terminal errors you did. I don't believe this is a bug however. The Prusa locks down when bed leveling fails and it tells you on the LCD screen that the printer must be reset. I assume serial communication is shut down by the printer at this point. There is nothing OctoPrint can do about that.

foosel, 2018-07-09:
If the printer triggers an `Error` and even state that it's internal `kill()` got called, that's the firmware telling OctoPrint that something just went seriously FUBAR and some manual intervention (= power cycling) by the user is necessary.

Unless you've explicitly disabled it, such errors from the printer will trigger a notication on the UI (in red even), but that's about all that OctoPrint can do here.

So from my understanding, this is expected behaviour.

tedder, 2018-07-09:
Thanks Gina. "expected behavior" is what I wanted to know from you.

closed 2018-07-09 by tedder