OctoPrint/OctoPrint #2722: Get octoprint running on python3

razerraz, 2018-07-09:
Maybe it's not right place for that, feel free to move/delete this
I manage to get octoprint running on python 3.6
For sure, it's far from running completly, but the core is loading, connection with printer ok, flask pages and sockjs push ok, base plugins ok, wizard ok.
It's mainly type conversion hacks, import changes and fighting with unmaintained sockjs-tornado that seems to be the main migration issue here. Nevertheless socket is partialy running too, with my own hacks in the module. I just have issues with crashs in the asyncio loop without any log at all, as I have with other asyncio projects.
It's mainly a game for me and a way to learn things about consequent basecode migration, but maybe my work can help here. Feel free to request all my changes via a fork or a pull request, keeping in mind that the merge will be a mess
Cheers

tedder, 2018-07-09:
See #1416 .

razerraz, 2018-07-09:
Well, for what I've seen so far, if the idea behind is to make something working both in python 2.7 & 3.x, the task will be very hard and bothering.
Perhaps having a python3 specific branch is the best way to go, imho, then I can do some PR

foosel, 2018-07-13:
I took the liberty to turn this into a Brainstorming ticket.

A python 3 specific branch is out of the question I fear. I already have way too much overhead as is, I seriously don't want to have to start doing everything twice across not only one but up to three concurrent development branches (`devel`, `maintenance`, `staging/maintenance`). 

The goal is to make this whole code base compatible to 2 and 3, and vendor bundle and adjust any third party dependencies that don't support this if push comes to shove. 

sockjs-tornado will already be vendor bundled starting with 1.3.9 anyhow since it appears to be unmaintained and there are some bug fixes available that are still not merged to the official release version, so I went that way.

The thing I'm most worried about myself is indeed unicode vs str and that will probably take the longest time to get right.

tedder, 2018-07-13:
Yeah, the str/bytes thing ends up being chased a lot. `2to3` isn't the prettiest for converting either.

Do you plan on cutting over or supporting both?

foosel, 2018-07-13:
I'll have to support both for a while until all the plugin authors have had a chance to convert their plugins as well. Another issue is finding a migration path for all those ancient OctoPi installs out there that will have to be switched to Python 3 as well then. Although i'm leaning towards "upgrade your images please" in that case.

tedder, 2018-07-13:
I'd prefer a short "support both" period- maybe only 1.5, and then 1.6 fully deprecates it. It was annoying to write Ansible plugins that had to support everything from py 2.6 to 3.x.

Just my vote, though.

foosel, 2018-07-13:
The current goal is to get both working with 1.4.0 and then drop support for 2 in 1.5.0. Maybe even declare that 2.0 due to semver vs plugins. I'm not fully sure about that yet.

I'm also not looking forward to this at all, especially since I already have enough on my plate as is, but well, no choice. Big bang would be easiest for *me* but suck for most users.

razerraz, 2018-07-13:
I have time right now and continue to sort everything out running on py3, without really thinking deeply on having the thing py2 compliant as well. Will come...
Right now, it's running on a venv, **all tests are passing ok**, just some curious things with decorators and functools. 
Of course, the list of changes is huge and affects sources and tests, but a lot of things can be done right now with very limited risk : 
- Use int argument for yaml indent attribute
- Wrap every map/filter/... into a list object
- Use six facilities for all isinstance() involving unicode

I will do one proper PR for every items mentioned, and one another fork with all my changes, the best horror movie you can watch I guess

foosel, 2018-07-13:
> Use six facilities for all isinstance() involving unicode

Careful, there's already a dependency on python-future for stuff like this.

smurfix, 2019-01-24:
@razerraz are you still working on porting to Py3?

razerraz, 2019-01-24:
@smurfix right now I'm quite busy with other things, but I will soon hopefully
There is a branch here : https://github.com/razerraz/OctoPrint/tree/Py3_devel that is running on py2 + py3 if I remember well, and if it can help right now

foosel, 2019-01-25:
There's also a ton of PRs open regarding Py3 support that I really need to get around to review and merge. Regular maintenance and life just keep getting in the way 😒

smurfix, 2019-01-25:
Yeah, life's like that. I just noticed these PRs … I started merging @razerraz 's branch up, can probably finish that this weekend. Then I could do the same thing with @ByReaL 's PRs, merge them all into a different branch, and cross-check the two. That should help.

ByReaL, 2019-01-25:
i'll after you merge mines in a branch i need probably few days to do some further testing and changesi would like to install that branch on py 2 and 3 and start an actual 3D print and play with the UI a bit

smurfix, 2019-01-27:
I'm working my way through the difference between @razerraz 's patches, @ByReaL 's patches, and @foosel 's maintenance work. Will finish tomorrow (probably) and then do some light testing.

smurfix, 2019-01-28:
I'm through with the basic integration stuff, though I'm sure I missed a few things (all still untested).

My branch is at github.com/smurfix/OctoPrint (py23 branch) though I'll probably rebase it to keep the changes reasonably clean and separate. I'll report back when I'm reasonably done with that, most likely this evening.

smurfix, 2019-01-28:
OK, tests for Python 2.7 and 3.6 pass. This is not done (I didn't yet try to print anything!) but I will no longer rebase the branch.

@foosel Please take a look. Maybe we could continue this in the main repository.

smurfix, 2019-01-28:
NB: I probably made some mistakes while cleaning up the merge conflicts, so my next task will be to merge @foosel 's maint and dev branches (unless she beats me to it … hint, hint ;-)  and then go through the diff from that to py23 with a hopefully-sufficiently-fine comb.

smurfix, 2019-01-28:
Yes, I merged all of the PRs. There were some conflicts, which is why IMHO this kind of work is best done in one line of development with one type of change per commit, and frequent rebasing - because you're sure to have missed something (which is what I did after mostly-octopus-merging all the py23 pull requests).

https://github.com/smurfix/OctoPrint/tree/py23 is correct. Yes, we do need a compatibility level thing, preferably one that we can read without importing the module in question. I'd use simple (i.e. ' eval' able) `OCTOPRINT_MIN_REL` and `OCTOPRINT_MAX_REL` integers, or maybe semver-style triples. Then again, if even Firefox with its insanely complex interface doesn't need more than a simple integer, then so do we. ;-)

There have been reasonably few `list(map())` instances. Again, I'm sure I've missed a few … 

WRT dependencies: Four or five of then required an uptick, according to @razerraz . See the current version of `setup.py` in my repository. I'll clean that up with another change shortly, as requiring different dependencies depending on Py2 vs. Py3 doesn't seem particularly maintainance friendly to me; there's the "future" module which is Py2 only, but *that's it* IMHO.

foosel, 2019-01-28:
> Yes, we do need a compatibility level thing, preferably one that we can read without importing the module in question.

I extract the name and version from installed plugins without actually importing or otherwise running them in any way through the AST of the module (I introduced that after I caught some plugins happily jumping into action just by importing them, allowing them to partially bypass manual disabling or blacklist). I guess something similar could be done here, with an assumed default of "py27 only" if not overwritten by the plugin author.

But I was actually talking about the repository and plugin manager as a first step, where we already check against OctoPrint versions & underlying operating system before an installation is even attempted let alone the plugin detected and its own metadata fetched.

smurfix, 2019-01-28:
Status: I went over the source with a fine-toothed comb and `list()`ed all instances of map/filter/keys/values/items that needed it (and un-`list()`ed a bunch that did not).

However, currently the server goes into an endless loop when I access the start page, for both Py2 and Py3. I suspect that I broke something when I merged the dev and main branches.

@foosel :: could you please merge maint into dev and get it to work? I will then do something sensible with the result so that (a) it's Py3 compatible and (b) the history isn't excessively convoluted, given the fact that I already did an obviously-somewhat-broken merge of these two branches while merging the Py3 PRs.

foosel, 2019-01-29:
@smurfix on it

foosel, 2019-01-30:
@smurfix Merge is complete and working in principle, however I see message duplication on the websocket and am not yet sure where that is coming from. Have to investigate further, might take until tomorrow. I don't want to push the current semi broken state since I know that a (very small) number of people use `devel`.

smurfix, 2019-01-30:
So push it to another branch …

foosel, 2019-01-31:
See https://github.com/foosel/OctoPrint/tree/dev/merge_201901

foosel, 2019-01-31:
Strike that, finally found and fixed the duplication bug, merge result is now available on `devel`

smurfix, 2019-01-31:
I still have a problem: With an empty .octoprint and a newly-minted Python2.7 virtualenv, the browser gets into an endless loop where it briefly shows the main screen, then the spinner, repeat ad infinitum.

This makes testing the Py3 version a bit difficult – it does the same thing, but that's hardly helpful at this point …
```
2019-01-31 20:21:52,690 - octoprint.server.util.sockjs - INFO - New connection from client: 10.107.2.5
2019-01-31 20:21:52,903 - octoprint.server.util.flask - INFO - Passively logging in user None from 10.107.2.5
2019-01-31 20:21:53,894 - octoprint.server.util.sockjs - INFO - Client connection closed: 10.107.2.5
2019-01-31 20:21:54,968 - tornado.access - WARNING - 403 GET /api/timelapse?unrendered=true (10.107.2.5) 3.08ms
2019-01-31 20:21:56,976 - octoprint.server.util.sockjs - INFO - New connection from client: 10.107.2.5
2019-01-31 20:21:57,023 - octoprint.server.util.flask - INFO - Passively logging in user None from 10.107.2.5
2019-01-31 20:21:58,037 - octoprint.server.util.sockjs - INFO - Client connection closed: 10.107.2.5
```
(repeats)

Browser console (copy+paste):
```
Navigated to http://dev.vm.smurf.noris.de:50080/#temp
User logged out packed_core.js:3286:25
... binding done packed_core.js:17214:13
Application startup complete packed_core.js:17226:13
Less has finished and no sheets were loaded. less.min.js:13:12757
Password fields present on an insecure (http://) page. This is a security risk that allows user login credentials to be stolen.[Learn More]
dev.vm.smurf.noris.de:50080
Password fields present on an insecure (http://) page. This is a security risk that allows user login credentials to be stolen.[Learn More] dev.vm.smurf.noris.de:50080
Starting dependency resolution... packed_core.js:16853:9
... dependency resolution done packed_core.js:16963:9
Initial application setup done, connecting to server... packed_core.js:17265:9
Connected to the server packed_core.js:14953:9
Finalizing application startup packed_core.js:17243:13
Going to bind 39 view models... packed_core.js:17122:13
Did not bind view model SoftwareUpdateViewModel to target #softwareupdate_confirmation_dialog since it does not exist packed_core.js:17175:33
Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate since it does not exist
```

foosel, 2019-01-31:
Huh... I'll take a look tomorrow, maybe the empty `.octoprint` folder is the key.

ByReaL, 2019-01-31:
Will try to have a look this weekend also.

Thanks,
Mircea Dan

foosel, 2019-02-01:
Good news, with an empty `.octoprint` folder I can reproduce this looping error.

foosel, 2019-02-01:
Solved it by removing the forcelogin plugin. We don't need it on `devel` because its function is pretty much superseded by the granular permission system, and its logic that's still based on the `maintenance` application initialization was causing the loop. Then I fixed the wizard which was broken by a misinterpretation on how deprecation wrapping in parent classes works. And then I fixed a bug in `maintenance` that caused the webcam to not be displayed and merged that into `devel` too. And now I hope the worst problems should be fixed. Phew.

foosel, 2019-02-01:
I think at this point the history around this point in time is pretty much 😲 anyhow, and it wouldn't be the first time a `git bisect` can run into such situations in this project. So I'd say: merge. Or rebase if that's an option. But I wouldn't go through the hassle of splitting everything up yet again.

Also: get well soon! And I'm frankly astonished that you feel capable of merging horror in that state, when I'm sick these days I try to not touch any code anymore, past experience has shown me that it rarely ends well 😅

smurfix, 2019-02-01:
You missed something:
`git grep DummyUser`

smurfix, 2019-02-01:
anyway, disregarding that, the current py23 branch (which I just pushed), while looking reasonably sane when looking at the diff, doesn't work with Py3 yet. Some more debugging and cleanup is necessary.

I'm of two minds about `from future import unicode_literals`ing the whole code base, but on the whole I think that we might progress faster if we do that.

smurfix, 2019-02-03:
Did that. Also did a bit of code cleanup. Initial startup now works, I also tested uploading. I didn't yet test actual printing, that'll happen tomorrow.

smurfix, 2019-02-04:
Branch status: communication with the printer works. I didn't actually print anything yet and will undoubtedly hit a few unicode-vs-bytes corners that have been overlooked, but I consider the work to be 95% finished and there should be no major issues left.

foosel, 2019-02-04:
> You missed something:
> `git grep DummyUser`

Should be fixed now. 

> I'm of two minds about `from future import unicode_literals`ing the whole code base, but on the whole I think that we might progress faster if we do that.

I fear you are right. I don't like it very much, but on the other hand it should indeed allow for faster progress. 

> I consider the work to be 95% finished and there should be no major issues left.

Sounds good! 👍 When you think it feels ready, can you create a consolidated PR? I'll then give that a test drive against a printer as well and if things look good we can merge (and if not, fix stuff). And then I have to think of a good strategy to avoid reintroducing incompatibility problems by merges from maintenance or other branches.

And a huge thank you for putting in that work, that really put a huge load off my shoulders!

smurfix, 2019-02-04:
I'll proceed until I have a working test print; should be tomorrow at the latest.

We probably should create a test case that actually runs a server and uploads+prints a test file to the virtual printer. That should catch many regressions.

smurfix, 2019-02-04:
See #3032.

smurfix, 2019-02-04:
OK. As the last step (so far …) I ran flake8 on he Py23 branch, which found a bunch of interesting errors – I recommend to add a run of "flake8 --ignore=D,E,W src tests setup.py" to the testsuite.

Printing and manual controls works for me. I found a problem with retransmission that I *think* is fixed (can' reproduce because my printer isn't *that* flaky). Somebody needs to run tests with an SD-card-equipped printer; mine isn't and that's the main part that might require some more work.

Barring strong opinions from @foosel 😇 I declare this branch to be in go-forth-and-get-it-merged quality.

smurfix, 2019-02-05:
> One question, in the PR you mentioned flake8 now running through (with the mentioned ignores) - I guess it would also make sense to just add that to .travis.yml to run on every build going forward, any objections to that?

No, of course not. Do it.

NB: I did a coupe of virtual printer tests but the emulated SD card directory was empty. My bug, your bug, or not implemented?

foosel, 2019-02-05:
> My bug, your bug, or not implemented?

Last I checked it did work here. I also have files in there (some I streamed, some I copied manually).

smurfix, 2019-02-05:
OK, I'll look into it.

foosel, 2019-02-06:
Print under py3 went through without hiccups, print under py2 is currently running and so far looking normal too. So the most important stuff appears to be working. Unless anything horrible gets unearthed by someone over night, I'll merge #3032 tomorrow.

@ByReaL you wanted a heads-up about this, this is the heads-up.

ByReaL, 2019-02-06:
thank you very much

razerraz, 2019-02-10:
I'm currently testing devel branch with a py3 venv : seems to work fine, what a tremendous job you've done on a small amount of time it's amazing. The speed gain is noticeable on the small nanopi board I'm using.
My test case is trying to install third party plugins on a fresh install, and I've already found a bug that prevent installing them, and sadly one of my PR is the cause. Fix can be found here :
https://github.com/razerraz/OctoPrint/commit/d269c361a4e4f90fbb35afa4948f8e88fe5fecd0

Otherwise, I ran into issues involving software update plugin, permissions, json encoding and got lost. The main exception was ::
```
Feb 09 17:55:34 3dprinter octoprint[4761]: [2019-02-09 17:55:34,119] ERROR in app: Exception on /plugin/softwareupdate/check [GET]
Feb 09 17:55:34 3dprinter octoprint[4761]: Traceback (most recent call last):
Feb 09 17:55:34 3dprinter octoprint[4761]:   File "/home/razer/OctoPrint-3c5aa60/venv/lib/python3.7/site-packages/OctoPrint-1.4.0.dev2000+g3c5aa60d-py3.7.egg/octoprint/server/util/flask.py", line 1567, in default
Feb 09 17:55:34 3dprinter octoprint[4761]:     return JsonEncoding.encode(obj)
Feb 09 17:55:34 3dprinter octoprint[4761]:   File "/home/razer/OctoPrint-3c5aa60/venv/lib/python3.7/site-packages/OctoPrint-1.4.0.dev2000+g3c5aa60d-py3.7.egg/octoprint/util/json/__init__.py", line 31, in encode
Feb 09 17:55:34 3dprinter octoprint[4761]:     raise TypeError
Feb 09 17:55:34 3dprinter octoprint[4761]: TypeError
Feb 09 17:55:34 3dprinter octoprint[4761]: During handling of the above exception, another exception occurred:
Feb 09 17:55:34 3dprinter octoprint[4761]: Traceback (most recent call last):
Feb 09 17:55:34 3dprinter octoprint[4761]:   File "/home/razer/OctoPrint-3c5aa60/venv/lib/python3.7/site-packages/Flask-0.12.4-py3.7.egg/flask/app.py", line 1982, in wsgi_app
Feb 09 17:55:34 3dprinter octoprint[4761]:     response = self.full_dispatch_request()
Feb 09 17:55:34 3dprinter octoprint[4761]:   File "/home/razer/OctoPrint-3c5aa60/venv/lib/python3.7/site-packages/Flask-0.12.4-py3.7.egg/flask/app.py", line 1614, in full_dispatch_request
Feb 09 17:55:34 3dprinter octoprint[4761]:     rv = self.handle_user_exception(e)
Feb 09 17:55:34 3dprinter octoprint[4761]:   File "/home/razer/OctoPrint-3c5aa60/venv/lib/python3.7/site-packages/Flask-0.12.4-py3.7.egg/flask/app.py", line 1517, in handle_user_exception
Feb 09 17:55:34 3dprinter octoprint[4761]:     reraise(exc_type, exc_value, tb)
Feb 09 17:55:34 3dprinter octoprint[4761]:   File "/home/razer/OctoPrint-3c5aa60/venv/lib/python3.7/site-packages/Flask-0.12.4-py3.7.egg/flask/_compat.py", line 33, in reraise
Feb 09 17:55:34 3dprinter octoprint[4761]:     raise value
Feb 09 17:55:34 3dprinter octoprint[4761]:   File "/home/razer/OctoPrint-3c5aa60/venv/lib/python3.7/site-packages/Flask-0.12.4-py3.7.egg/flask/app.py", line 1612, in full_dispatch_request
Feb 09 17:55:34 3dprinter octoprint[4761]:     rv = self.dispatch_request()
Feb 09 17:55:34 3dprinter octoprint[4761]:   File "/home/razer/OctoPrint-3c5aa60/venv/lib/python3.7/site-packages/Flask-0.12.4-py3.7.egg/flask/app.py", line 1598, in dispatch_request
Feb 09 17:55:34 3dprinter octoprint[4761]:     return self.view_functions[rule.endpoint](**req.view_args)
Feb 09 17:55:34 3dprinter octoprint[4761]:   File "/home/razer/OctoPrint-3c5aa60/venv/lib/python3.7/site-packages/OctoPrint-1.4.0.dev2000+g3c5aa60d-py3.7.egg/octoprint/server/util/flask.py", line 1242, in decorated_view
Feb 09 17:55:34 3dprinter octoprint[4761]:     return func(*args, **kwargs)
Feb 09 17:55:34 3dprinter octoprint[4761]:   File "/home/razer/OctoPrint-3c5aa60/venv/lib/python3.7/site-packages/OctoPrint-1.4.0.dev2000+g3c5aa60d-py3.7.egg/octoprint/access/permissions.py", line 158, in _decorated
Feb 09 17:55:34 3dprinter octoprint[4761]:     rv = f(*args, **kw)
Feb 09 17:55:34 3dprinter octoprint[4761]:   File "/home/razer/OctoPrint-3c5aa60/venv/lib/python3.7/site-packages/OctoPrint-1.4.0.dev2000+g3c5aa60d-py3.7.egg/octoprint/plugins/softwareupdate/__init__.py", line 586, in check_for_update
Feb 09 17:55:34 3dprinter octoprint[4761]:     unless=lambda: force)(view)()
Feb 09 17:55:34 3dprinter octoprint[4761]:   File "/home/razer/OctoPrint-3c5aa60/venv/lib/python3.7/site-packages/OctoPrint-1.4.0.dev2000+g3c5aa60d-py3.7.egg/octoprint/server/util/flask.py", line 1089, in decorated_function
Feb 09 17:55:34 3dprinter octoprint[4761]:     response = f(*args, **kwargs)
Feb 09 17:55:34 3dprinter octoprint[4761]:   File "/home/razer/OctoPrint-3c5aa60/venv/lib/python3.7/site-packages/OctoPrint-1.4.0.dev2000+g3c5aa60d-py3.7.egg/octoprint/plugins/softwareupdate/__init__.py", line 544, in view
Feb 09 17:55:34 3dprinter octoprint[4761]:     dict(name=gettext("setuptools"), current=self._environment_versions.get("setuptools", "unknown"), minimum=MINIMUM_SETUPTOOLS)])))
Feb 09 17:55:34 3dprinter octoprint[4761]:   File "/home/razer/OctoPrint-3c5aa60/venv/lib/python3.7/site-packages/OctoPrint-1.4.0.dev2000+g3c5aa60d-py3.7.egg/octoprint/server/util/flask.py", line 277, in fixed_jsonify
Feb 09 17:55:34 3dprinter octoprint[4761]:     dumps(data, indent=indent, separators=separators) + '\n',
Feb 09 17:55:34 3dprinter octoprint[4761]:   File "/home/razer/OctoPrint-3c5aa60/venv/lib/python3.7/site-packages/Flask-0.12.4-py3.7.egg/flask/json.py", line 167, in dumps
Feb 09 17:55:34 3dprinter octoprint[4761]:     rv = _json.dumps(obj, **kwargs)
[... cut ...]
Feb 10 09:22:17 3dprinter octoprint[1297]: OrderedDict([(<class 'frozendict.frozendict'>, <function <lambda> at 0xb2cac5d0>), (<class 'octoprint.access.users.User'>, <function Server.run.<locals>.<lambda> at 0xb2920030>), (<class 'octoprint.access.groups.Group'>, <function Server.run.<locals>.<lambda> at 0xb29200c0>), (<class 'octoprint.access.permissions.OctoPrintPermission'>, <function Server.run.<locals>.<lambda> at 0xb2920108>)])::ffff:192.168.3.253 - - [10/Feb/2019 09:21:54] "GET /sockjs/info?t=1549786914779 HTTP/1.1" 404 -
```

I feel something unusual in this list without ability to understand what's going on here

Cheers

foosel, 2019-02-12:
@razerraz Did you run into this softwareupdate issue just while checking for updates? I haven't yet encountered it myself, so I wonder what made it pop up on your end.

And thanks for the heads-up regarding razerraz@d269c36

razerraz, 2019-02-12:
Well I can't still reproduce it either with all third party plugin disabled. I was guessing having done the same before reporting, right now I doubt about it.
There are still a lot of problems with third parties, mainly import stuff and of course, unicode mess... I've already done PRs for some I'm using, but obviously I miss something.

foosel, 2019-02-13:
The plugin situation is going to be a fun one for the foreseeable future. I'm actually pondering not only introducing a flag in the plugin repository but a new [control property](http://docs.octoprint.org/en/master/plugins/controlproperties.html) to signal py3 compatibility during the dual support time.

smurfix, 2019-02-13:
Or you could require `__plugin_check__` to return something other than `True` (3, for instance 😄) when called from Python3.

razerraz, 2019-02-13:
Maybe the occasion to think about something like a minimal test tool for plugins developers. When I ran into troubles developing one, this idea comes already to my mind...

foosel, 2019-02-13:
What are you envisioning there?

razerraz, 2019-02-13:
Nothing right now, since I'm quite busy trying to resolve the issue I mention here, and sadly comes from one of my plugin.
I can't understand why my own updater script doesn't want to work : 
`    def get_update_information(self):
        return dict(ABL_Expert=dict(
            displayName='ABL Expert Plugin',
            displayVersion=self._plugin_version,
            current=self._plugin_version,
            type='python_checker',
            python_checker=ABL_VersionCheck,
            pip=('https://framagit.org/razer/Octoprint_ABL_Expert/-/archive' +
                 '/master/Octoprint_ABL_Expert-master.zip')))`

`class ABL_VersionCheck:
    @classmethod
    def get_latest(cls, target, check, full_data=False, online=True):
        current_version = check.get("current", "1.0.0")
        remote_version = current_version
        if online:
            r = requests.get('https://framagit.org/razer/Octoprint_ABL_Expert'
                             + '/raw/master/VERSION?inline=false')
            if r.status_code == 200:
                remote_version = r.text.strip().encode('ascii', 'ignore')

        info = dict(local=dict(name=current_version, value=current_version),
                    remote=dict(name=remote_version, value=remote_version))

        return info, remote_version == current_version`

Maybe not the right place for that, but this was working before with py2

foosel, 2019-02-13:
I'll throw it into a test plugin and see if I can reproduce this. Also just fixed an evil windows-only issue caused by an actual CPython bug, fun times.

foosel, 2019-02-13:
The problem is that your plugin returns it's latest version as a `bytes` object, and the json encoder in Py3 dosn't like that. I've pushed the above commits to allow JSON encoding of `bytes` (by turning them into utf-8 strings) but I'm not sure if that is really the best approach. Happy to discuss.

smurfix, 2019-02-13:
*its* latest version 😛

I can't think of a better solution either, other than telling people "don't use bytes for version strings dammit" which may or may not be a better solution.

razerraz, 2019-02-13:
Yeah I just got it on the way for my son's art class, something wrong with the dep in the middle : request...
Just got time to solve the issue before reading you
thanks

razerraz, 2019-02-13:
> I'll throw it into a test plugin and see if I can reproduce this. Also just fixed an evil windows-only issue caused by an actual CPython bug, fun times.

by the way, what a funny idea some have to use windows for managing a 3dprinter. Some even use it on a raspberry pi. What a pity...

jneilliii, 2019-02-13:
Yeah, I've been watching this thread waiting for the plugin developer call out.  I'm scared honestly. Would love to have some sort of flag to tell me my stuff is going to break.

foosel, 2019-02-14:
> by the way, what a funny idea some have to use windows for managing a 3dprinter.

The next thing might scar you for life: I develop this under Windows 😅 Otherwise I wouldn't notice such things, most people run it under Linux:

[image]

> _its_ latest version 😛

I know, I know... typed too fast, didn't notice until now :P

>  I'm scared honestly. Would love to have some sort of flag to tell me my stuff is going to break.

We are rather thinking about a flag to not have your plugin load unless you have tested it and know it *won't* break. 

I'm in the middle of merging `devel` onto the comm refactoring branch and turning that Py3 compatible as well. When I'm through with that (or when I *desperately* need a break from it to keep the growing frustration at bay - my money is on this coming first...) I'll try a first throw at a py compat control property. Overloading `__plugin_check__` is something I'd rather avoid and instead introduce an actual flag, since I can evaluate a flag via the AST without even having to import the plugin. The earlier we can detect incompatibility, the better.

[21 more replies]

closed 2020-03-04 by foosel