OctoPrint/OctoPrint #2724: Long pre heat seems to cause printer to skip steps

PrestonL, 2018-07-10:
#### What were you doing?

Printing with Xinkebot Orca 2 Cygnus Dual Extruder Printer with Heated Bed.

When dealing with a moderate or long heat up time, the printer seems to end up skipping the earlier steps of the gcode.  I noticed that while the printer is heating up, octoprint is still sending it commands and steps.  I'm guessing the buffer in the printer is overflowing and dropping off the previous commands?  Can Octoprint be configured to wait on sending the next steps until the preheat is completed?

1. I create a gcode file in slic3r
2. Sending it directly from slic3r or uploading the gcode file from the web interface causes the same problem described above
3. Putting the same file on an SD Card does not yield the same result

#### What did you expect to happen?

Expected the printer to wait until the preheat was complete to start sending the commands for the printer to start doing things.

#### What happened instead?

The print head never went to zero and started moving slowly to the X/Y position of the part, as if it has been resumed from a pause, then it started acting like it was printing the middle of the print job by printing in mid air.

#### Did the same happen when running OctoPrint in safe mode?

Yes, the data collected and posted in this report is from OctoPrint running in safe mode.

#### Version of OctoPrint

1.3.8

#### Operating System running OctoPrint

OctoPi 0.15.1

#### Printer model & used firmware incl. version

Xinkebot Orca 2 Cygnus

Firmware Version 1.3.17

#### Browser and version of browser, operating system running browser

macOS 10.13.5 (17F77), Safari 11.1.1 (13605.2.8)

#### Link to octoprint.log

https://gist.github.com/PrestonL/20db5c8a7bab820173d4795bee45a00c

#### Link to contents of terminal tab or serial.log

https://gist.github.com/PrestonL/fec2df4cbfbbfe6b99ae0eca73243ed9

#### Link to contents of Javascript console in the browser

(No output)

#### Screenshot(s)/video(s) showing the problem:

[image]

Result with manual pre heat

[image]

Result without manual pre heat

I have read the FAQ.

foosel, 2018-07-13:
This looks like it runs the same kind of [ridiculously](https://github.com/foosel/OctoPrint/issues/2309#issuecomment-356893095) [broken](https://github.com/foosel/OctoPrint/issues/2309#issuecomment-356922357) [ass](https://github.com/foosel/OctoPrint/issues/2309#issuecomment-357183123) [firmware](https://github.com/foosel/OctoPrint/issues/2309#issuecomment-357283302) as the Anycubic Mega i3, down to the `M115` responses  and the completely non standard resend behaviour that I've already had to dissect in #2309. I suggest you try the two plugins [linked here](https://github.com/foosel/OctoPrint/issues/2309#issuecomment-358713557) and see if they might also fix the issues with your printer.

If not I'd suggest to scream at your printer's vendor since they completely f#cked up their firmware's behaviour.

PrestonL, 2019-05-23:
[image]
[image]

I did want to post an update and mention that I did end up talking to the makers of the firmware for this board (a 32-bit one made by Chitu3D).  They were not interested in correcting their firmware, so I was wondering if there were any other solutions to this parsing issue.  I tried the patches to no avail, and while the graph shows the temperature and target temperatures are correct, the UI under it does not.

I'm guessing octoprint needs to wait to send print commands until it's preheated, otherwise it overruns the buffer and when it does start printing it will skip over the gcode it had been feeding it.

My solution so far has been to simply preheat before printing, but it would help a lot if I could just hit print and walk away. I have my settings tuned in to the point where the printer can run jobs almost 100% hands free.

Any ideas on better workarounds?

foosel, 2019-05-24:
> They were not interested in correcting their firmware, 

🙄

> so I was wondering if there were any other solutions to this parsing issue.

Yes, a third plugin that throws the target temperature on `T0` away and uses `T` instead. That would need to be done on line receiving though.

> I tried the patches to no avail

Would be interesting what you expected to happen and what happened instead ;) It should have solved the issue with the `wait` wrongly used instead of the `busy` (and hence made OctoPrint stop sending commands to the printer while it's happily sending `wait`), and stripping the `M110` parameters. 

> the graph shows the temperature and target temperatures are correct, the UI under it does not

We are talking about the wrong target temperature on `T0` here? In that case I disagree. The graph and the UI underneath show the same thing in the legend and my guess is that the faded red line of `T0` is just covered by the faded blue of the bed and hence doesn't show up.

> I'm guessing octoprint needs to wait to send print commands until it's preheated

Which it should do with the `rewrite_wait_to_busy` plugin. If that doesn't work, either the plugin wasn't installed correctly or for some reason doesn't "catch" here and then it needs to be investigated why.

PrestonL, 2019-05-24:
> 🙄

That was my reaction as well

> Yes, a third plugin that throws the target temperature on T0 away and uses T instead. That would need to be done on line receiving though.

Does this plugin exist somewhere?

I'll go back and retry this with the other plugins not installed and installed, and see what shows differently.  It's been a while since I've originally visited this issue.

foosel, 2019-05-24:
> Does this plugin exist somewhere?

Not yet. 

> I'll go back and retry this with the other plugins not installed and installed, and see what shows differently.

The thing is, the log will be the same, but the behaviour should get fixed with the plugins installed.

PrestonL, 2019-05-26:
It appears octopi now works as intended with the "rewrite_wait_to_busy" plugin.  I had it installed but not enabled (and I don't think the permissions were right at first either, I can't remember now).

Also, the temperature seems to be reading correctly now somehow?!

[image]

[image]

The -45C is because there is no extruder/sensor plugged into the second spot (this printer allows you to swap single/dual extruders).

I'll experiment and look into the temperature sensor issue.  Would it be beneficial for me to add these findings and steps to the wiki here? https://github.com/foosel/OctoPrint/wiki/Supported-Printers

foosel, 2019-05-27:
> It appears octopi now works as intended with the "rewrite_wait_to_busy" plugin

OctoPrint ;) [OctoPi is just the image and has nothing to do with actually driving your printer](https://community.octoprint.org/t/what-is-the-difference-between-octoprint-and-octopi-are-they-the-same-thing/185).

> Would it be beneficial for me to add these findings and steps to the wiki here? https://github.com/foosel/OctoPrint/wiki/Supported-Printers

Yes, definitely!

foosel, 2019-10-31:
Closing this since I don't see anything that I still need to do here.

closed 2019-10-31 by foosel