OctoPrint/OctoPrint #2727: BlueprintPlugin mixin paths/urls?

paukstelis, 2018-07-11:
#### What were you doing?
I have multiple octoprint instances setup with redirected/reverse proxy to allow external authentication.
I was trying to setup Octolapse, but found that all of the POSTS from the BlueprintPlugin mixin were not going to the correct URL, leading to countless errors.
The base url for the octoprint instances would be: `http://something.something/octoprint1`, but everything from this mixin was going to: `http://something.something/plugin` which obviously doesn't exist. I would expect these POSTS to go to: `http://something.something/octoprint1/plugin`

#### What did you expect to happen?
Expected base url to be correct
#### What happened instead?
404
#### Did the same happen when running OctoPrint in safe mode?
Did not try, but shouldn't matter

#### Version of OctoPrint
1.3.8-1.3.9rc2

#### Operating System running OctoPrint
Linux

I have read the FAQ.

foosel, 2018-07-11:
Did you tell OctoPrint about the prefixes via the `X-Script-Name` header?

See also [here](https://discourse.octoprint.org/t/reverse-proxy-configuration-examples/1107)

paukstelis, 2018-07-11:
Totally missed that. Thanks. I will confirm it works as expected and then close.

paukstelis, 2018-07-11:
It was inconsistency of trailing forward slash in apache set up. Closing.

closed 2018-07-11 by paukstelis