OctoPrint/OctoPrint #2728: Print freezes minutes after browser is closed than resumes

odudex, 2018-07-12:
#### What were you doing?
Printing a file via octoprint, USB connected to a MKS Sbase running Marlin 2

1. Turn on printer and octoprint
2. Upload gcode file using browser
3. Start to print
4. Close the browser or Octoprint tab (tried closing directly and logging out first)
5. Wait a moment (1 to 5 minutes)
6. Print pauses for ~2seconds causing a blob in the part
7. Print resumes. Everything keeps working fine.
8. If you open and close Octprint in browser the pause will happen again a few minutes later

It is happening all times I test it, in PC and phone

#### What did you expect to happen?
I expected to run the print without pauses that cause artifacts in parts
#### What happened instead?
The pauses that happen minutes after browser is closed cause artifacts in parts
#### Did the same happen when running OctoPrint in safe mode?
I did no run in safe mode.

#### Version of OctoPrint
1.3.8

#### Operating System running OctoPrint
ARMBIAN 5.38 stable Debian GNU/Linux 9 (stretch) 4.14.18-sunxi

#### Printer model & used firmware incl. version
MKS Sbase running Marlin 2.0.x

#### Browser and version of browser, operating system running browser
Chrome 67.0.3396.99 - windows 10 updated AND Chrome - android updated

#### Link to octoprint.log
[octoprint.log](https://github.com/foosel/OctoPrint/files/2189266/octoprint.log)

I have read the FAQ.

foosel, 2018-07-12:
Please test if this also happens in safe mode.

odudex, 2018-07-12:
I ran it in safe mode.
First it asked to update. I did it, but despite the fact I track the Release channel it installed the "1.3.8.post1.dev0+gedd00cfa" version. Is it a dev version?
After that I rebooted at safe mode again, and it asked to update again, but this time i didn´t accepted.
I printed an object in safe mode and the bug did NOT happen.
I than rebooted again in default mode. It did not ask for updates, and for my surprise said I'm using the 1.3.8 stable version.
The good news is that the bug is GONE! Everything ran smoothly.

I'm sorry about taking your time and I hope you get something useful from this case. Maybe there is something happening when checking update in safe mode.

Thanks,

foosel, 2018-07-13:
Hm... ok. Closing this then since it's no longer happening. Maybe some files got mixed up somewhere, which might also explain your weird version reports. `.dev0` hints at OctoPrint *thinking* it was running something not from the `master` branch (see also ["How OctoPrint is versioned" in the Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-octoprint-is-versioned)), but the commit at the end indicates you indeed were. I have no idea what happened there, but since everything is running now I'll just assume a temporary glitch in the matrix.

closed 2018-07-13 by foosel